bk1285/rpi_wordclock #188: Seconds Plugin

R0lleston, 2021-07-18:
I'm having an issue with the seconds plugin, the 10s are displaying, however the 1's are not - all that I see beside the 10's are two or three individual pixels that flash on and off in a pattern.

Now I see the code for the seconds plugin, references a german wordclock and I'm guessing thats the problem (my native language is english). What I need to ascertain is, what I need to modify to get the seconds to correctly work based on an English wordclock. Cheers

R0lleston, 2021-07-19:
[image]

plotaBot, 2021-07-19:
First of all: Welcome to the group of proud owners of a wordclock and I apreciate your interest in the seconds plugin!

I think what you mean are the first lines from time_seconds.py:
[image]
Unfortunatelly this is a relic from copying the basecode from time_german.py and after coding my things into it I must have forgotten to change this part...
To cut things short: As long as you keep the original matrix dimensions, the code will work on any language you have selected.
Codewise there is no explanation for this behaviour for now. 
Maybe there are some unknown interferences with your hardware when the LEDs are updated at a higher rate. 
Greetings

R0lleston, 2021-07-19:
Hi Plotabot, thanks for the reply and clarification. All default plugins are working correctly without any display errors so it is indeed a mystery. The only thing I can say is that pixels for the 'ones' do come on in a repeating pattern (where the word five appears). With the version of the code being the phenze python 3. Though I doubt that would be the issue. The mystery continues

R0lleston, 2021-07-23:
@plotaBot is your python3 variant of wordclock usable or still in development? If usable, are there specific steps to follow to install and/or specific version of the OS ?

R0lleston, 2021-07-31:
Ok, seemed to have it working now. It appears to be an issue with @phenze version (specifically the time_seconds.py file. I copied the code from @bk1285 development of time_seconds and it worked correctly. It also fixed an issue with @bk1285 code where the 1s would just from 1 to 3.
So to sum it up
I kept the @phenze plugin.py py3 file as is, but replaced his time_seconds.py with @bk1285 py3 version

closed 2021-07-31 by R0lleston