bk1285/rpi_wordclock #190: Feature request : scrolling message

R0lleston, 2021-07-23:
Anyone up for creating a continuous scrolling message plugin?
- message could be saved in config file
- option of single colour or rainbow

Example could be "Happy Birthday Monty"

bk1285, 2021-07-24:
Nice idea, @R0lleston 

If anyone want's to go for it, please

* base your work on current develop branch (python3)
* consider a plugin (combined with message within the config file) or even a API-endpoint, allowing to send text via the webinterface to the clock.

R0lleston, 2021-07-25:
@bk1285 look forward to seeing what your team of coding magicians can come up with.......  my kids would love a feature like this for welcoming guests to say one of their birthday parties, especially if its colourful

FrankX0, 2021-07-26:
Like the idea!
I can see two implementations:
1. Special occasions.
Using a plugin, with API-endpoint. Setting the text, repeat time, colour, etc. through the webinterface
2. Repeating occasions.
Like for birthdays, new-years, etc.. These should go into the config file with text, date, repeat time, etc.

R0lleston, 2021-07-26:
@FrankX0 sounds perfect, we just need to talk nicely to a coder to bring this to reality.
how about also the option (if possible) of the background pixel lit slightly, with the words themselves fully lit (to the give the effect of one of those scrolling message devices).

bk1285, 2021-07-27:
Sound's promising, @FrankX0. Amendment to your comment: 
Option 2. should probably use the new transitions, where the text (including colour-settings, etc.) is animated without explicitly running a plugin. Transitions would need some further refactoring first.
Option 2 could furthermore be realized:
* 2.a: Based on config file
* 2.b: Accessing a calendar via caldav (.... if anyone volunteers... 🙈 ) At least, in my calendar, all birthdays are noted and additional (reoccurring) events could be displayed as some sort of reminder... 😉

R0lleston, 2021-08-14:
@FrankX0 @bk1285 has anyone been allocated to this or taken it on themselves to have a go

FrankX0, 2021-08-16:
I will start having a look (especially the web interface) when I get back from holiday.
Cannot make any promises though.

bk1285, 2021-08-16:
Sounds great, @FrankX0!

as mentioned before, please use the latest develop branch of this repo to start implementing this. (...since this project keeps receiving pull requests from branches, which were forked loong time ago... )

Best,
Bernd

FrankX0, 2021-08-23:
Initial prototype for the user interface:

- Text to scroll
- Start date and time
- Repeating period
- Enable/disable switch

[image]

Feedback is appreciated.

R0lleston, 2021-08-23:
@FrankX0 Big thumbs up from here, looks great. Is it envisaged that the message will be able to scroll continiously and with time delays of seconds as well as minutes? If so, then brilliant !

FrankX0, 2021-08-24:
@R0lleston If the repeat time is set to 0 it will scroll continuously (maybe I should call it delay?).
I was thinking specifying the delay in minutes was small enough.
To not over-complicate things, I could change the delay to be specified in seconds. Would that be OK?

R0lleston, 2021-08-24:
@FrankX0 that would be perfect (delay in seconds) fantastic stuff

bk1285, 2021-08-25:
Looks great, @FrankX0 

How about hiding this in a collapsed layout by default?

Best,
Bernd

R0lleston, 2021-10-10:
@bk1285 @FrankX0 I realise life is busy but was wondering how progress is going :) hope the weekend is going well

R0lleston, 2021-10-11:
Fantastic news @FrankX0 , appreciate all that you are doing :)

FrankX0, 2021-10-19:
I have made an initial implementation, with the user interface but yet without the repeating message.
You can find it here:
https://github.com/FrankX0/rpi_wordclock/tree/develop
It is based on the current (Pyhton3) development branch.
The message will show when the "enable" switch is turned on (although I could not test it on an actual clock).

I am not yet sure how to do the time triggering and repeating of the text though.
I was thinking of using a [scheduler](https://github.com/dbader/schedule), but that's not working yet.
@bk1285 any suggestions? I can certainly use some help in this area.

FrankX0, 2021-10-31:
Hi @R0lleston
I just created a pull request (#210) for the development branch.
Please test and let me know what you think.

R0lleston, 2021-10-31:
> Hi @R0lleston I just created a pull request (#210) for the development branch. Please test and let me know what you think.

@FrankX0 i saw this just pop up in my email. Very exciting, I'll try and a squeeze some time in over the weekend for a test. Thank you :)

R0lleston, 2021-11-01:
@FrankX0 getting the following error when trying to run

[image]

FrankX0, 2021-11-01:
Sorry, last minute change.
Should be fixed now.

R0lleston, 2021-11-02:
@FrankX0 have reinstalled from your 'develop' directory, no errors now, but the config file (or other bits) need modifying to allow for the new plugin......  ie it doesnt show in the list to enable or disable, thus doesnt come up as an option on your web interface

FrankX0, 2021-11-02:
This is not a new plugin. So the config file does not need modifying.
It could be that your browser is still caching the old web interface.
In (Windows) Chrome you can do an "Empty Cache and Hard Reload".
On iOS I always have problems when changing the web interface. This will "correct itself eventually".

R0lleston, 2021-11-02:
@FrankX0 you were bang on the money, it was an iOS issue, I retried lastnight on my windows pc and it worked nicely. I do like the clock implementation you did. I had a play around and it worked nicely, the only thing I noticed is that if you have message running and you change it to another, the display glitches with random lit up LEDs briefly and then recovers and works correctly. Without knowing a hell of a lot about how the clock works, it could be just how the clock software operates and your new code has highlighted it. But really good job.... hopefully a few others will give this a good test.

R0lleston, 2021-11-04:
@bk1285 @FrankX0 I'm not sure how must testing you two have done on an actual clock, but there is an issue with how the scrolling word feature when existing plugins are in use. This morning while demonstrating it to my kids - I had the matrix - with time plugin going, when the scrolling time starts, the matrix keeps going, while the text scroll begins to start up and these is random leds coming on. We might need to clear the wordclock display of the current plugin, before starting the scrolling text. Its easily reproducable if you want to see what im referring to :)

FrankX0, 2021-11-05:
For now this feature is implemented to only work the standard clock-face, not with any other plugins. Sorry.
When I have time I can see how to support other plugins.
But for now, please only use it with the standard face.

FrankX0, 2021-11-14:
@R0lleston I just made a fix for the scrolling text in combination with the matrix-with-time plugin.
Can you try again?