bk1285/rpi_wordclock #191: Feature Request: Debounce on buttons

R0lleston, 2021-07-26:
@bk1285 my project uses both physical buttons and the webapp. I've noticed that when going through the menu with the buttons: the following has been observed :
1) trying to select the menu item the clock jumps into and back out of the plugin
2) selecting doesnt always select first time.

A common programming feature when using physical buttons is a debounce, whereby we have small filter (delay) implemented when a button is pressed to prevent spurious activations due to the switches mechanism. 

Not sure if this has been implemented, but if it has the delay might need increasing a bit

mrksngl, 2021-07-26:
Actually, there is debouncing.

It's not implemented by the wordclock code itself, but by the GPIO library (some hardware is even able to do that on its own, so the library might just pass it to the kernel).

Long story short:
https://github.com/bk1285/rpi_wordclock/blob/8347da70e6dfa6da182d0d32bfca681a320f4728/wordclock_interfaces/gpio_interface.py#L42

the `bouncetime` parameter (ms) in those 3 calls does the job. But yes, it should be configurable.

R0lleston, 2021-07-26:
@mrksngl thanks for the headsup :)

closed 2021-07-26 by R0lleston