bk1285/rpi_wordclock #194: Suggestions Installation Instructions - development & eventually the new master

R0lleston, 2021-07-29:
@bk1285 a suggestion based on attempting to get your 'development' python3 version of the wordclock working
1) state the version of raspbian that you used to get the code working. a) stretch appears to work, b) buster seemed to bring up issues I couldnt find fix (right at the very end when attempting to run wordclock.py for the 1st time)
2) state the requirements of python 3 version. e.g. I attempted to get the clock working via py3.6.14 under stretch, but Astral2.2 wouldnt install (it couldnt find v2.2 despite it the owner of astral stating py3.6 was compatible), V3.7 under buster worked but I had Buster issues as referenced in 1 above
3) when I got the development version installed, and ran wordclock.py for the 1st time a few error messages popped up which were addressed by googling the issue (and corrected by installing a couple of extra items). Maybe these sorts of issues could be noted in the install info.
4) I had to google how to install a different branch from master, so it might be useful for people to mention how to do this

I guess my suggestion would be to just expand the installation info out a bit (not necessarily dumb it down), but fill in the gaps so its clear to a new user what they need to do and in what order.
e.g.
a) install OS version xyz
b) install python version 3.xyz
c) do this
d) do that etc

Just little things to give the user the confidence that when they follow the steps, they'll get a consistant end result regardless of their technical expertise. 

Just as a side note - while using your dev version, the time in seconds plug in would skip numbers consistantly eg it would show 11sec, then 13 sec, 15s then 17s so there is a little bit of code that needs looking at.

Keep up the great work

phenze, 2021-07-30:
i have done this in my fork:

https://github.com/phenze/rpi_wordclock/tree/master-python3

### Python3
When using the Python 3 branch the follwing changes have to be made during installation.
Tested on latest Raspian Buster (02.12.2020)

#### System Packages
```
sudo apt-get install python3-pip python3-scipy scons git swig fonts-freefont-ttf python3-rpi.gpio
```

#### Python Packages
Instead of *flask-restplus* which is no longer maintained we are using now *flask-restx*.
For the great *rpi-ws281x* Library there is no need to compile it yourself. It is just installed via pip now.

```
sudo pip3 install pytz astral feedparser pillow svgwrite freetype-py netifaces monotonic flask-restx rpi-ws281x
```

R0lleston, 2021-07-30:
@phenze I'm running your version because it seems a much more easier install.
Some things I've noticed tho
1) your time as seconds plug in is buggy (ones dont display correctly, where Bernds version works)
2) weather plugin doesnt load
3) sunrise and sunset plugin isnt working right . Displays same incorrect times whatever country you enter

bk1285, 2021-08-09:
Hi all,

thanks for the updates! I considered your comments and optimized/minimized the old outdated documentation. Latest version (using python3) can be found here: https://rpi-wordclock.readthedocs.io/en/develop/doc_software_setup.html

Any feedback welcome. Esp. with respect to missing dependencies, errors, etc. during setup. As soon as this is confirmed to be working with py3, I'll merge it to master.

bk1285, 2021-08-09:
... refering to the comment above, please open an issue, if you run into problems while using latest documentation on develop.

Best,
Bernd

closed 2021-08-09 by bk1285