bk1285/rpi_wordclock #196: Inconsistent time between modules

R0lleston, 2021-08-04:
Not sure what is going on here, but I've noticed that at times the 'the display time in words' plugin is one hour ahead of the time_default plugin (which reads correctly). It seems to be a common bug as I've noted it in @bk1285 development version and @phenze python3 master version.

It doesnt occur all the time, which made me think that when I translated the wording from german to english. To confirm, in the  time_as_german plugin should the hours list be ? 
TWELVE
ONE
TWO
THREE
FOUR
FIVE
SIX
SEVEN
EIGHT
NINE
TEN
ELEVEN
TWELVE 
?

R0lleston, 2021-08-04:
As a follow up, I have a video of this happening if it is helpful

bk1285, 2021-08-09:
Thanks for reporting, @R0lleston Seems to be related to pythons "new" division: https://www.informit.com/articles/article.aspx?p=1439189 ...which was an issue for the default plugins as well.

bk1285, 2021-08-09:
https://github.com/bk1285/rpi_wordclock/commit/2cfccb3565c833802996ba65bb25aafebca1f303 should fix it. Please reopen, if this is not the case.

Best,
Bernd

closed 2021-08-09 by bk1285