bk1285/rpi_wordclock #197: Question Stencilsize

Silence667, 2021-08-04:
I have print the scelton from file with my 3D-printer. Now i just want to make a version of the stencil. In Inkscape the stencil is just 450mm and the scelton just have the size of 500mm. Need i to change the size and how could i do that? Thanks for your help.

bk1285, 2021-08-09:
Hi @Silence667,

when increasing the skeleton size, keep the spacing of the LEDs on the LED-strip in mind: Their spacing must equal (or bigger) than the spacing of your reflectors.

Concerning the stencil: You probably have to adopt it to your custom configuration on your own. Maybe you can use the script create_layout.py within this repository (which reads all according values like spacing, border size, etc. from the wordclock_config.cfg.

Best,
Bernd

bk1285, 2021-08-09:
Please reopen, in case you don't find any suitable solution.

closed 2021-08-09 by bk1285

Silence667, 2022-04-19:
Ich mache hier nochmal die Frage auf. Also ich habe jetzt die Clock soweit zusammengebaut und auch (bis auf die Schalter und den Lichtsensor (Amazon lässt auf sich warten) verbaut. Jetzt habe ich im Moment weiterhin das Problem, dass ich nicht genau weiß wie ich die Front konstruiere und in welchem Versand ich diese erstellen könnte. Ich habe schon gegoogelt, aber vielleicht bringen mich Erfahrungsberichte zur Front und kleine Tipps weiter. 
Danke schonmal für die Hilfe und Respekt für den großartigen Code.