bk1285/rpi_wordclock #199: Seconds plugin jumps numbers (bernd development version)

R0lleston, 2021-08-09:
This could be similar issue to one that Bernd solved recently. But the seconds count plugin jumps number eg 31, 33, 34, 35, 36,38 (it's always jumps two positions). I haven't checked for time accuracy but it's something that needs looking at

bk1285, 2021-08-09:
Hi @R0lleston, I tried to reproduce this on latest develop, but there are no jumps on my clock. Do you make any specific changes apart from latest develop?

Best,
Bernd

R0lleston, 2021-08-09:
@bk1285 no changes at all. Happy to record a video of it an email it through to you

R0lleston, 2021-08-09:
Uploading 20210809_215514.mp4…

R0lleston, 2021-08-09:
https://user-images.githubusercontent.com/87622394/128692946-bab039ce-602e-4807-bca4-a05251f672db.mp4

bk1285, 2021-08-09:
I see your point. Is it possible, that you have accidentally two instances of the software running? Use `pgrep python` via command line to determine the number of active python processes (should be one).

bk1285, 2021-08-09:
Hi @R0lleston please reopen, if your issue persists..

Best,
Bernd

closed 2021-08-09 by bk1285

R0lleston, 2021-08-10:
[image]
That's from the clock being turned on from cold. @bk1285 how do we fix this ?

R0lleston, 2021-08-10:
[image]
@bk1285 it seems to be coming from wordclock.py

R0lleston, 2021-08-14:
@bk1285 strangely enough this issue seems to have sorted itself out using @phenze development version using alot of your code.