bk1285/rpi_wordclock #202: PIP cant find latest packages

R0lleston, 2021-08-10:
Can someone advise why when I try and install the below (using sudo pip3 install) I get errors finding some of the package versions. EG with astral - it cant find 2.2 and brings up a list of available old versions in red test - from the command line.

Same occurs with feedparser 6.05, Flask 2.0.1 - at that point I gave up.

astral==2.2
coloredlogs==15.0.1
feedparser==6.0.5
flask-restplus==0.13.0
Flask==2.0.1
freetype-py==2.2.0
netifaces==0.11.0
monotonic==1.6
pillow==8.2.0
rpi-ws281x==4.3.0

For reference, I'm running the latest stretch lite version, with python 3.8.11 installed on a Rpi Zero W

phenze, 2021-08-11:
Why you are running stretch version ?

Just use the current buster version. I have tried it yesterday and everything works like a charm.
https://www.raspberrypi.org/software/operating-systems/

For further install instructions take a look here (a have made some changes during installation):
https://github.com/bk1285/rpi_wordclock/pull/201

R0lleston, 2021-08-11:
@phenze I've tried loading your new version on buster and it installs without issue. However on running wordclock.py im getting 'no option language in section : plugin_time_default any ideas

phenze, 2021-08-11:
Yes.
You have to define language again in plugin_time_default section ! That is because there are many more languages for time_default than for stencil.

I have splitted that in my version so that i can use bavarian language.

```
[plugin_time_default]
activate = True
language = bavarian
```

closed 2021-08-11 by R0lleston