bk1285/rpi_wordclock #203: Suggestion - Second plugin

R0lleston, 2021-08-14:
Is there a way to disable the fading effect when the numbers tick over, as it gives the illusion that the seconds are counting faster than they are. Would be much nicer to have a straight on and off (like a digital watch) or an option for fading or on /off

phenze, 2021-08-16:
Just comment out the next two lines under the comment "previous seconds, dimming down"  in time_in_seconds/plugin.py

```
# previous seconds, dimming down
# taw_indices = self.taw.get_time(currentSecond - 1 if currentSecond != 0 else 59)
# wcd.setColorBy1DCoordinates(taw_indices, wcc.Color(i, i, i))
```

R0lleston, 2021-08-16:
@phenze thank you,  worked perfectly  :)

closed 2021-08-16 by R0lleston