bk1285/rpi_wordclock #209: Raspberry Pi Zero 2 W: RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

kas1kas, 2021-10-31:
I tried to run the software (python3 version thanks to Phenze) on the Zero 2 W. This should work, why not?
But I get this message:
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 187, in <module>
    word_clock = wordclock()
  File "/home/pi/rpi_wordclock/wordclock.py", line 67, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 70, in __init__
    self.strip.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault

The exact same software runs fine on Pi 3B+.
Anyone knows what this error message means?
thanks, Kas

R0lleston, 2021-10-31:
I using the exact same piece of hardware on my clock - the Pi Zero and used @bk1285 Development version - Python 3 variant, if you follow the instructions carefully you shouldnt have any problems. Have you tried wiping the sd card and starting from scratch ?

FrankX0, 2021-10-31:
@R0lleston: are you using a Pi Zero or a Pi Zero **2**?
From the traceback messages I conclude that the rpi_ws281x package needs to be updated to support the new hardware.

R0lleston, 2021-10-31:
> @R0lleston: are you using a Pi Zero or a Pi Zero **2**? From the traceback messages I conclude that the rpi_ws281x package needs to be updated to support the new hardware.

@FrankX0 Ah, i didnt read the post correctly, yes, I'm using the Pi Zero, not the 2. Good spotting

kas1kas, 2021-11-02:
@FrankX0 Thanks for the tip. Right now I'm on a trip for a week, but I'll post the results here once I'm back.
If someone else gets it to run on the Pi Zero 2 W before me, with the new library, please respond.

bk1285, 2021-11-02:
Hi @kas1kas, I agree with @FrankX0, that it's probably worth trying to update rpi-ws281x. Please report back, if the issue is resolved. Meanwhile, I leave this open.

Best,
Bernd

kas1kas, 2021-11-09:
Here is my update:
$**sudo pip3 install rpi-ws281x**
**requirement already satisfied: rpi-ws281x in usr/local/lib/python3.7/dist-packages (4.3.0)**

I performed
**sudo pip3 install --force-reinstall rpi-ws281x** 
This did upgrade to the newer version 4.3.1

now everything works!

closed 2022-02-09 by kas1kas