bk1285/rpi_wordclock #211: Issues to install guided by the documentation

SimJaaenis, 2021-11-03:
First of all, thanks for the great project. I built a clock a few years ago and have often been asked about it. I've also given some to family and friends for Christmas. The hardware components seem to have remained the same since then. But the SD card has gone over time and I wanted to reload the software now. Unfortunately I failed. In the meantime, there are already inquiries from my gift recipients and it becomes more critical.
I tried several times to install the software according to the current documentation and failed. I can't get it to run. 
I tried different software distributions of Raspian (Stretch/Buster, with and without GUI), the fork of phenze (including the autoinstaller), the developer version, try to install the pip requirements myself. All attempts failed somewhere. In 2015 it was just copy and paste.

What are the hardware (RasPi revision) & software (Raspberry Pi OS version) requirements?
The idea of the autoinstaller I find very good for fools like me, if it would work up to date.
Maybe you find time to take a blank micro SD card, install the Raspberry Pi OS without desktop, and then try the documentation yourself. Try this 10 times and you will have my last 2 days.

coolio210, 2021-11-11:
Thank you for asking. I've got exactly the same problems. 
I hope there's someone who is able to explain the solution to our problem.

phenze, 2021-11-12:
Have you tried my install instructions here ?

https://github.com/phenze/rpi_wordclock/tree/master-python3

I dont use the auto installer. So it is not tested in my fork.
I have used Buster without gui and everything works fine ;) 

Please specify your your problem a bit more detailed. In which step of the installation you have problems and what are the exact error message.

phenze, 2021-11-12:
Take care to use the python3 commands.

It is `pip3 install` and not `pip install`.
And the software is started with `python3` instead of `python`.

Please also tell us which python version you are working with:
`python3 --version`

If you want to use Bernds Development Branch (which you should prefer instead of mine) a detailed install instruction you can find here:

https://github.com/bk1285/rpi_wordclock/tree/develop

Beside of using the python 3 commands another important step is to checkout the python3 branch:
`git checkout --track origin/develop`

coolio210, 2021-11-15:
Hello, thank you for answering.
My python3 version is 3.7.3.
python3 scipy 1.1.0-7, scons 3.0.1-2, swig 3.0.12-2, python3-pip
18.1-5+rpt1

I tried to install Bernd's Development Branch. After trying to
install the repository I get the message "Bitte committen oder stashen Sie
Ihre Änderungen, bevor Sie Branches wechseln"

Unfortunately I haven't got a clue, what this means.

phenze, 2021-11-16:
This means that you have made changes in a wrong file.

You can habe a look with `git status` which files you have changed.
I am quite sure that is is the config file. Please make sure to make a copy of the example.config file (without the example in the name) and not directly change the example file.

You can reset your changes with 
`git reset --hard`

After that the change to the development branch should work again :
`git checkout --track origin/develop`

coolio210, 2021-11-17:
Hi,

I had so many problems so I deleted the whole directory and installed
everything new like shown here:

https://github.com/bk1285/rpi_wordclock/tree/develop

The whole installation finished without an error-message. (Without the part
„Brightness Sensor“).

But when I try to start the program with the command sudo python3
wordclock.py  I get the response:

File "wordclock.py", line 2, in <module>
    import coloredlogs
ModuleNotFoundError: No module named ‚coloredlogs'

In my view there’s something missing ?!

or Android

FrankX0, 2021-11-17:
You are correct: one package is missing.
Please do the following: `sudo pip3 install coloredlogs`
Then try again.

coolio210, 2021-11-17:
OK - after installing the missing „colored logs“ I got a new error after trying to run 

sudo python3 wordclock.py:

Traceback (most recent call last):
  File "wordclock.py", line 195, in <module>
    word_clock = wordclock()
  File "wordclock.py", line 34, in __init__
    self.config = wccfg.wordclock_config(self.basePath)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 9, in __init__
    self.loadConfig(basePath)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 20, in loadConfig
    copyfile(pathToReferenceConfigFile, pathToConfigFile)
NameError: name 'copyfile' is not defined

FrankX0, 2021-11-17:
Just added a fix for that: https://github.com/bk1285/rpi_wordclock/pull/210/commits/67c5ceedbb305cbe0a6b7402aeada645b7935659

Hang in there: I think you should almost be there.

coolio210, 2021-11-17:
> Just added a fix for that: 67c5cee <https://github.com/bk1285/rpi_wordclock/commit/67c5ceedbb305cbe0a6b7402aeada645b7935659>
> Hang in there: I think you should almost be there.

Hi, 

I’m on my way …
I just added 

from shutil import copyfile

to the file word clock_config.py and it took a little bit longer till the next error messages appeared - I think that’s a good sign ;-) 

Unfortunately there are many errors: 

Traceback (most recent call last):
  File "/usr/lib/python3.7/configparser.py", line 788, in get
    value = d[option]
  File "/usr/lib/python3.7/collections/__init__.py", line 914, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.7/collections/__init__.py", line 906, in __missing__
    raise KeyError(key)
KeyError: 'language'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 36, in request
    return getattr(self.config, method)(*args)
  File "/usr/lib/python3.7/configparser.py", line 791, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'language' in section: 'plugin_time_default'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/configparser.py", line 788, in get
    value = d[option]
  File "/usr/lib/python3.7/collections/__init__.py", line 914, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.7/collections/__init__.py", line 906, in __missing__
    raise KeyError(key)
KeyError: 'language'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "wordclock.py", line 195, in <module>
    word_clock = wordclock()
  File "wordclock.py", line 50, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 113, in __init__
    language = ''.join(config.get('plugin_time_default', 'language'))
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 48, in get
    return self.request("get", *args)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 39, in request
    return getattr(self.reference_config, method)(*args)
  File "/usr/lib/python3.7/configparser.py", line 791, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'language' in section: 'plugin_time_default'

phenze, 2021-11-18:
This exceptions are all the same. Like the error message says "No option 'language' in section: 'plugin_time_default'".

Try adding the language option under the plugin_time_default section in your config file:
`language = german` for example.

@bk1285 this is still a problem when you dont want to use german in both sections (wordclock_display and plugin_time_default). In my case i have to use

```
[wordclock_display]
language = german
```
and 
```
[plugin_time_default]
language = bavarian
```
for example. We should add this option in the config reference file.

coolio210, 2021-11-20:
Hi, it’s me again.

I added the line 

language= german

in the file wordclock_config.cfg in the directory /rpi_wordclock/wordclock_config

I added it underneath the definitions of the other languages.
Now i get the following error-messages: 

2021-11-20 10:39:06 raspberrypi root[978] WARNING Defaulting to reference value for [plugin_time_default] language
Traceback (most recent call last):
  File "/usr/lib/python3.7/configparser.py", line 788, in get
    value = d[option]
  File "/usr/lib/python3.7/collections/__init__.py", line 914, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.7/collections/__init__.py", line 906, in __missing__
    raise KeyError(key)
KeyError: 'language'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 36, in request
    return getattr(self.config, method)(*args)
  File "/usr/lib/python3.7/configparser.py", line 791, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'language' in section: 'plugin_time_default'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.7/configparser.py", line 788, in get
    value = d[option]
  File "/usr/lib/python3.7/collections/__init__.py", line 914, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.7/collections/__init__.py", line 906, in __missing__
    raise KeyError(key)
KeyError: 'language'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "wordclock.py", line 195, in <module>
    word_clock = wordclock()
  File "wordclock.py", line 50, in __init__
    self.wcd = wcd.wordclock_display(self.config, self.wci)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_display.py", line 113, in __init__
    language = ''.join(config.get('plugin_time_default', 'language'))
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 48, in get
    return self.request("get", *args)
  File "/home/pi/rpi_wordclock/wordclock_tools/wordclock_config.py", line 39, in request
    return getattr(self.reference_config, method)(*args)
  File "/usr/lib/python3.7/configparser.py", line 791, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'language' in section: ‚plugin_time_default'

Do I have to change or add the line somewhere else?

FrankX0, 2021-11-20:
I'm a bit puzzled (like you I guess).
I cannot reproduce your error with the current develop branch.
There is no key `language` required in the `plugin_time_default` section in the config file (like the error suggests).

Can you maybe clone the current develop branch again?

phenze, 2021-11-22:
Haha i get it! 
He is using my development branch because in the documentation of bernds repo there is the wrong path.

In my development branch you need the `language` key under `plugin_time_default`.
@coolio210 like i mentioned you should put the key in the `plugin_time_default` section and not underneath the language definitions.
Your error message are exactly the same as in the first post so obviously your changes didn't do anything.

But anyway. You should use bernds version.

Try the following steps:
```
cd ~
rm -Rf ~/rpi_wordclock
git clone https://github.com/bk1285/rpi_wordclock.git
cd rpi_wordclock/
git checkout --track origin/develop
```

FrankX0, 2021-11-22:
Try the following: `sudo pip3 install configparser coloredlogs`

coolio210, 2021-11-23:
Hello, here's the result:
sudo pip3 install configparser coloredlogs
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting configparser
  Downloading
https://www.piwheels.org/simple/configparser/configparser-5.1.0-py3-none-any.whl
(19 kB)
Requirement already satisfied: coloredlogs in
/usr/local/lib/python3.9/dist-packages (15.0.1)
Requirement already satisfied: humanfriendly>=9.1 in
/usr/local/lib/python3.9/dist-packages (from coloredlogs) (10.0)
Installing collected packages: configparser
Successfully installed configparser-5.1.0
***@***.***:~ $ cd rpi_wordclock
***@***.***:~/rpi_wordclock $ sudo python3 wordclock.py

*Traceback (most recent call last): File
"/home/pi/rpi_wordclock/wordclock.py", line 1, in <module>    import
ConfigParserModuleNotFoundError: No module named 'ConfigParser'*

Sorry for causing so much trouble. 😞😞😞

phenze, 2021-11-23:
You have said:
```
and as recomended *Bernds Version*
git clone https://github.com/bk1285/rpi_wordclock.git
```

Did you also switch to the develop branch (`git checkout --track origin/develop`) ?
It looks like you are working on the master (python2) branch.

When you are in the `rpi_wordclock` directory what is the output of `git status` ?

coolio210, 2021-11-23:
it says:

Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

nichts zu committen, Arbeitsverzeichnis unverändert

phenze, 2021-11-23:
So jetzt auf Deutsch vll hilft das ja was ;) 

Du musst auf den develop branch wechseln. origin/master ist der alte Stand ohne die python3 Änderungen.

Dazu wechselt du in das rpi_wordclock Verzeichnis
`cd rpi_wordclock/`

Und anschließend musst du den Branch mit folgendem Befehl wechseln
`git checkout --track origin/develop`.

Danach sollte `git status` folgendes ausgeben:
`Ihr Branch ist auf demselben Stand wie 'origin/develop'.`

Dann sollte das alles laufen.
Und wie gesagt bitte etwas in die Materie einlesen 👍   Wir helfen hier sehr gerne aber können keine Schulung in Umgang mit git/terminal etc geben.

coolio210, 2021-11-24:
Hello,

I finally made it.
In case there are more  people who are not familiar with Linux and
programming like me, here's a short summary how I got it work:

1. install all packets like shown here:
https://github.com/bk1285/rpi_wordclock/tree/develop
2. add the line "from shutil import copyfile" in the file
wordclock_config.py (directory wordclock_tools)
3. add the line "language=german" in the section [plugin_time_default] in
the File wordclock_config.cfg (directory wordclock_config)
4. run the program with "sudo python3 wordclock.py"

Thank you so much for your patience and the essential hints.

bk1285, 2022-09-01:
latest main-branch now holds again a consistent documentation -- based on python3

closed 2022-09-01 by bk1285