bk1285/rpi_wordclock #214: Add Swedish as language in rpi wordclock

poa0314, 2021-11-30:
Everything here is new to me. I very much like to build a word clock in swedish. I have the electronics and the mechnical build in place but I struggle with the software that I am new to, but do my best to understand. From reading about others having different problems I have managed to ge the clock running in german. I thougt that the easiest way would be to overwrite german parts with the swedish equals. I have a layout for a frontplate in swedish, what else do I have to look in to? I would be very pleased if I could get any help.

FrankX0, 2021-12-13:
Do you still need help?
I would suggest to create a new `swedish.py` entry in `rpi_wordclock/wordclock_plugins/time_default/`
With the help of other languages and the [wiring layout](https://rpi-wordclock.readthedocs.io/en/master/_images/wiring_front_bernds_wiring.png) and your own front design, you can then convert words to LED positions in `swedish.py`,
And please use the `develop` branche. And once finished, make a pull request.

Please let me know if you need additional help.

poa0314, 2021-12-13:
Yes I still need help, and I am very pleased if you would help me.
I have the hardware under control but need support with software.
I have a layout for the frontplate and I think that I can look att german.py an make a swedish.py.
Could you please explain "use develop branche and make pull request". As you can understand from my comments I have a lot to learn, this is a new world to me. I have read the basics of python to try and understand the scripts, and I have also made some basic scripts to control a ledstrip.
I have spent a lot of time on this project and hope that I can finish it if I can get some support.

phenze, 2021-12-14:
Take a look here to understand how to use the development branch :
https://github.com/bk1285/rpi_wordclock/issues/211

For a quick overview how git works:
https://rogerdudler.github.io/git-guide/index.de.html

To make a pull request you have to do the following steps:
- Make a new branch with the development branch as a base
- commit your changes
- push the new branch to bernds repo
- make a pull request  (https://github.com/bk1285/rpi_wordclock/pulls)

Can you post a photo of your front layout? maybe i can help you.

phenze, 2021-12-14:
And of course we need all possible translations

* Klockan är fem över
* Klockan är tio över

etc for every 5 minutes steps.

poa0314, 2021-12-14:
I have created a time_swedish.py -file. I have one question regarding this file. In german the time end with "UHR" this is not used in swedish, can i only remove these rows or do I hav to do someting elese also?
I have also modified wordclock_config and added swedish in language_options.

[time_swedish.txt](https://github.com/bk1285/rpi_wordclock/files/7709984/time_swedish.txt)
[wordclock_config.txt](https://github.com/bk1285/rpi_wordclock/files/7710003/wordclock_config.txt)

I get this result when starting the clock, and it still run i german.

pi@raspberrypi:~ $ sudo python3 rpi_ws281x/python/rpi_wordclock/wordclock.py
Parsing /home/pi/rpi_ws281x/python/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to bavarian.
Imported plugin 0: "feed_parser".
Imported plugin 1: "time_as_words_dutch".
  INFO: No activate-flag set for plugin __pycache__ within the config-file. Will be imported.
No active exception to reraise
Failed to import plugin __pycache__!
Imported plugin 2: "test_strip".
Imported plugin 3: "matrix".
Imported plugin 4: "restart".
Imported plugin 5: "leds_off".
  INFO: No activate-flag set for plugin time_in_seconds within the config-file. Will be imported.
Imported plugin 6: "time_in_seconds".
Imported plugin 7: "time_as_words_german".
No module named 'pywapi'
Failed to import plugin weather_forecast!
Imported plugin 8: "tetris".
Imported plugin 9: "shutdown".
Imported plugin 10: "rainbow".
Not found brigtness sensor value 
Using brigtness sensor : False
  Selected "time_default" as default plugin
Imported plugin 11: "time_default".
Imported plugin 12: "sunrise".
  INFO: No activate-flag set for plugin snake within the config-file. Will be imported.
Imported plugin 13: "snake".
Imported plugin 14: "ip_address".
Imported plugin 15: "time_matrix".
Running plugin time_default.
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
^CTraceback (most recent call last):
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock.py", line 182, in <module>
    word_clock.run_forever()
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock.py", line 172, in run_forever
    self.run()
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock.py", line 144, in run
    self.runPlugin()
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock.py", line 121, in runPlugin
    self.plugins[self.plugin_index].run(self.wcd, self.wci)
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock_plugins/time_default/plugin.py", line 186, in run
    event = wci.waitForEvent(2)
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock_interfaces/event_handler.py", line 38, in waitForEvent
    self.__wait_for(lambda: self.event != self.EVENT_INVALID, seconds)
  File "/home/pi/rpi_ws281x/python/rpi_wordclock/wordclock_interfaces/event_handler.py", line 81, in __wait_for
    self.condition.wait(waittime)
  File "/usr/lib/python3.9/threading.py", line 316, in wait
    gotit = waiter.acquire(True, timeout)

poa0314, 2021-12-14:
This is the front layout in swedish

[Urtavla.pdf](https://github.com/bk1285/rpi_wordclock/files/7710224/Urtavla.pdf)

phenze, 2021-12-14:
`Setting language to bavarian.`
Are you sure you are on the correct repo ? Look like you are on mine.

What is the output of ?
`git status`

In your time_swedish you can modify what you want to. Removing the lines of code is fine.

To get the swedish language working you have to modify 
`wordclock_tools/wordclock_display.py`

Add an import for your language
`import wordclock_plugins.time_default.time_swedish as time_swedish`

Add an else statement for swedish (Line 114)
```
elif language == 'swedish':
    self.taw = time_swedish.time_swedish()
```

poa0314, 2021-12-14:
I have nov modified wordclock_display.py as you described.

git status gives the following:
----------------------------------------------------------
pi@raspberrypi:~/rpi_ws281x $ git status
På grenen master
Din gren är à jour med "origin/master".

Ospårade filer:
  (använd "git add <fil>..." för att ta med i det som ska checkas in)
	python/rpi_wordclock/

inget köat för incheckning, men ospårade filer finns (spåra med "git add")

If I stand in folder rpi_wordclock
-----------------------------------------------------
pi@raspberrypi:~/rpi_ws281x/python/rpi_wordclock $ git status
På grenen master-python3
Din gren är à jour med "origin/master-python3".

Ändringar ej i incheckningskön:
  (använd "git add <fil>..." för att uppdatera vad som ska checkas in)
  (använd "git restore <fil>..." för att förkasta ändringar i arbetskatalogen)
	ändrad:        wordclock_tools/wordclock_display.py

Ospårade filer:
  (använd "git add <fil>..." för att ta med i det som ska checkas in)
	wordclock_config/wordclock_config_ori.cfg
	wordclock_plugins/time_default/time_swedish.py
	wordclock_tools/wordclock_display_ori.py

inga ändringar att checka in (använd "git add" och/eller "git commit -a")
------------------------------------------------------------------
Do you want me to change to english or can you understand what i means?

If I start the clock it still works in german
---------------------------------------------------------
pi@raspberrypi:~ $ sudo python3 rpi_ws281x/python/rpi_wordclock/wordclock.py
Parsing /home/pi/rpi_ws281x/python/rpi_wordclock/wordclock_config/wordclock_config.cfg
Interface type set to gpio_low
Wiring configuration
  WCA_WIDTH: 11
  WCA_HEIGHT: 10
  Num of LEDs: 114
  Wiring layout: bernds_wiring
WARNING: Default brightness value not set in config-file: To do so, add a "brightness" between 1..255 to the [wordclock_display]-section.
  Setting language to bavarian.
Imported plugin 0: "feed_parser".
Imported plugin 1: "time_as_words_dutch".
  INFO: No activate-flag set for plugin __pycache__ within the config-file. Will be imported.
No active exception to reraise
Failed to import plugin __pycache__!
Imported plugin 2: "test_strip".
Imported plugin 3: "matrix".
Imported plugin 4: "restart".
Imported plugin 5: "leds_off".
  INFO: No activate-flag set for plugin time_in_seconds within the config-file. Will be imported.
Imported plugin 6: "time_in_seconds".
Imported plugin 7: "time_as_words_german".
No module named 'pywapi'
Failed to import plugin weather_forecast!
Imported plugin 8: "tetris".
Imported plugin 9: "shutdown".
Imported plugin 10: "rainbow".
Not found brigtness sensor value 
Using brigtness sensor : False
  Selected "time_default" as default plugin
Imported plugin 11: "time_default".
Imported plugin 12: "sunrise".
  INFO: No activate-flag set for plugin snake within the config-file. Will be imported.
Imported plugin 13: "snake".
Imported plugin 14: "ip_address".
Imported plugin 15: "time_matrix".
Running plugin time_default.
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
 * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)

phenze, 2021-12-14:
Okay there are several problems. I will try to explain it step by step.

**1. You dont need to checkout and compile the rpi_ws281x library**

Did you follow the python3 install instructions here ?
https://github.com/bk1285/rpi_wordclock/tree/develop

**2. The wordclock repo is in the wrong path (will be fixed in step 3)**
- Currently it is `~/rpi_ws281x/python/rpi_wordclock`
- It should be `~/rpi_wordclock`

**3. You have checked out the wrong repo as i thought**
Steps to fix that
```
cd ~
git clone https://github.com/bk1285/rpi_wordclock.git
cd rpi_wordclock/
git checkout --track origin/develop
```

**4. Before doing any changes**
- Copy config file `cp ~/rpi_wordclock/wordclock_config/wordclock_config.reference.cfg ~/rpi_wordclock/wordclock_config/wordclock_config.cfg `
- Try to start the wordclock now without any changes
- When everything works fine apply your changes again

**5. When noting helps**
- i have made the neccessary changes in my repo (except time_swedish changes)
```
cd ~
git clone https://github.com/phenze/rpi_wordclock rpi_wordclock_phenze
cd rpi_wordclock_phenze/
git checkout --track origin/feature/swedish
```
- Change time_swedish for your needs and send me the changes

poa0314, 2021-12-14:
I am very greatful that you take the time to help me.

I think I start from scratch on a new SD-card and follow your instructions.

Is it possible to attach a .py-file here, I had to make it a txt-file to be able to attach it

[time_swedish.txt](https://github.com/bk1285/rpi_wordclock/files/7714484/time_swedish.txt)

phenze, 2021-12-15:
a text file is okay.
You can post the code directly instead.

I have commited your time_swedish changes in my branch. Seems to work 👍 

When you re install everything you can checkout my swedish branch directly instead of bernds develop branch (Step 5).

Later you can switch to bernds version again when the changes are merged.

poa0314, 2021-12-15:
**I feel that the success is coming closer!**
I installed everything from scratch using https://github.com/bk1285/rpi_wordclock/tree/develop

After switching off audio it all worked fine i german.

**Then I:**
cd ~
git clone https://github.com/phenze/rpi_wordclock rpi_wordclock_phenze
cd rpi_wordclock_phenze/
git checkout --track origin/feature/swedish

**And I got the following result**
pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ git clone https://github.com/phenze/rpi_wordclock rpi_wordclock_phenze
Klonar till "rpi_wordclock_phenze"...
remote: Enumerating objects: 5145, done.
remote: Counting objects: 100% (2000/2000), done.
remote: Compressing objects: 100% (1638/1638), done.
remote: Total 5145 (delta 557), reused 1751 (delta 356), pack-reused 3145
Tar emot objekt: 100% (5145/5145), 15,99 MiB | 2,72 MiB/s, klart.
Analyserar delta: 100% (2559/2559), klart.
pi@raspberrypi:~ $ cd rpi_wordclock_phenze/
pi@raspberrypi:~/rpi_wordclock_phenze $ git checkout --track origin/feature/swedish
Grenen feature/swedish ställdes in att spåra fjärrgrenen "feature/swedish" från "origin".
Växlade till en ny gren "feature/swedish"
pi@raspberrypi:~/rpi_wordclock_phenze $ 

I guess I did something wrong since I now have two folders.
rpi_worldclock and worldclock_phenze

I understand that it must be frustrating trying to explain to a novice like me, how to perform the installation. My defence is that I am new to this and that I will be 70 in a few months and my brain perhaps works at a lower speed than earlier in life.
Thanks again for helping me!

phenze, 2021-12-15:
70 ??
And you doing that kind of stuff ?
### Stor respekt !

Everything is fine now. When the clock works in german that is exactly what we wanted.

You have two folder because you have used two different versions of the wordclock software.

In the folder `rpi_worldclock ` there is @bk1285's Version which you should use in future. When all the swedish changes are merged back.

But for the moment you can use the software installed in the `rpi_wordclock_phenze ` folder because there are all the needed changes for swedish (hopefully).

So all you have to do for now is:
- Change the config file for the installation in the folder `rpi_wordclock_phenze` (`rpi_wordclock_phenze/wordclock_config/wordclock.cfg`) to use swedish language
- Run the software in the `rpi_wordclock_phenze`folder with `sudo python3 /home/pi/rpi_wordclock_phenze/wordclock.py`

Your cronjob entry should then look like that:
`@reboot sudo python3 /home/pi/rpi_wordclock_phenze/wordclock.py`

poa0314, 2021-12-15:
Thank you for the encouraging comments!
These days I do whatever I like from morning to evening and  this is one of the things I like to spend my time on doing.

**The clock is now working in Swedish!!!**

I found some mistakes in time_swedish.py that i have corrected.
I add the corrected file at the end.

I try to understand how the plugins can be activated?

Is it trigged by the pushbuttons and if so is this the only way to run them?

Another question is how to connect to wifi if I make a clock to a friend. Can it be done from a PC or even a phone?

[time_swedish.txt](https://github.com/bk1285/rpi_wordclock/files/7720137/time_swedish.txt)

FrankX0, 2021-12-15:
The plugins can be activated by using the web interface.
Just go to the IP address assigned to your clock.

phenze, 2021-12-16:
Okay great. Is now everything working as expected for swedish? Then i will as @bk1285 to merge your changes to the main repo.

The Plugins can be activated in both ways (Buttons and via Web Interface).

**1. Buttons** 
- You can configure which the gpio pins for the buttons in the config file (wordclock_interface section)
- You can also define which state will trigger a button push (gpio_low or gpio_high --> that means if there is voltage when pressing the button or not).
- With the return/middle button you get to the main menu of the clock (white clock should be displayed)
- You can now select the plugins with the left/right buttons
- Another push on the return Button will start the selected plugin
- I don't use the buttons anymore so i hope this is correct

**2. Web Interface**
- A better way to control the clock is the web interface
- For this to work you need a working wifi connection on you raspberry
- Depending on the Raspberry model you either have a wifi chip onboard (since Raspberry 3) or you need a extra wifi dongle (edimax)
- Wifi can be configured via raspi-config
- Take a look here (Method 2: Configure WiFi from Command Line with Raspi-Config) - https://www.seeedstudio.com/blog/2021/01/25/three-methods-to-configure-raspberry-pi-wifi/)

poa0314, 2021-12-16:
**I have tried the Web interface and most of the plugins works great.**

The only real problem I have now is that the clock suddenly can switch to german layout and then later back again. It can also be flickering between the Swedish and German layouts.

I do not know if this gives you an idea of the problem.

pi@raspberrypi:~ $ sudo python3 rpi_wordclock_phenze/wordclock.py
2021-12-16 09:59:42 raspberrypi root[1256] INFO Software version: v1.0.0-dev-113-ga9b49b6
2021-12-16 09:59:42 raspberrypi root[1256] INFO Parsing /home/pi/rpi_wordclock_phenze/wordclock_config/wordclock_config.cfg
2021-12-16 09:59:42 raspberrypi root[1256] INFO Interface type set to gpio_low
2021-12-16 09:59:42 raspberrypi root[1256] INFO Wiring configuration
2021-12-16 09:59:42 raspberrypi root[1256] INFO   WCA_WIDTH: 11
2021-12-16 09:59:42 raspberrypi root[1256] INFO   WCA_HEIGHT: 10
2021-12-16 09:59:42 raspberrypi root[1256] INFO   Num of LEDs: 114
2021-12-16 09:59:42 raspberrypi root[1256] INFO   Wiring layout: bernds_wiring
2021-12-16 09:59:42 raspberrypi root[1256] INFO Setting language to swedish.
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 0: "feed_parser".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2021-12-16 09:59:42 raspberrypi root[1256] WARNING Defaulting to reference value for [plugin_test_strip] activate
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 1: "test_strip".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 2: "matrix".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 3: "restart".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 4: "leds_off".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 5: "time_in_seconds".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2021-12-16 09:59:42 raspberrypi root[1256] WARNING Assumes no temperature sensor to be attached.
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 6: "weather_forecast".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 7: "tetris".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 8: "shutdown".
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 9: "rainbow".
Using brigtness sensor : False
2021-12-16 09:59:42 raspberrypi root[1256] INFO Selected "time_default" as default plugin
2021-12-16 09:59:42 raspberrypi root[1256] INFO Imported plugin 10: "time_default".
2021-12-16 09:59:43 raspberrypi root[1256] INFO Imported plugin 11: "sunrise".
2021-12-16 09:59:43 raspberrypi root[1256] INFO Imported plugin 12: "snake".
2021-12-16 09:59:43 raspberrypi root[1256] INFO Imported plugin 13: "ip_address".
2021-12-16 09:59:43 raspberrypi root[1256] INFO Imported plugin 14: "time_matrix".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Exception ignored in thread started by: <bound method web_interface.threaded_app of <wordclock_interfaces.web_interface.web_interface object at 0x75478568>>
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock_interfaces/web_interface.py", line 71, in threaded_app
    self.app.run(host='0.0.0.0', port=port)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 990, in run
    run_simple(host, port, self, **options)
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 1052, in run_simple
    inner()
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 996, in inner
    srv = make_server(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 847, in make_server
    return ThreadedWSGIServer(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.9/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.9/http/server.py", line 138, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.9/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
2021-12-16 09:59:53 raspberrypi root[1256] INFO Running plugin time_default.

phenze, 2021-12-16:
Yes. 
There a running another instance of the wordclock.

Remember that you have to versions installed.
Did you add anything to the crontab yet  (`sudo crontab -e`)?

Remember in your case you have to change the line:
`@reboot sudo python3 /home/pi/rpi_wordclock/wordclock.py`
to
`@reboot sudo python3 /home/pi/rpi_wordclock_phenze/wordclock.py`

Change it and then after a restart everything should run fine.

When adding anything to the crontab you dont have to manually start the wordclock. This will get startet automatically

poa0314, 2021-12-16:
The plugins and the clock now works fine, but I found an error in time_swedish.
In german number ZWÖLF is on the first and last row of list(range).
I guess that it has to do with how the time is presented in german, and perhaps the use of "UHR" in the end. Do you have enough information to guide me how to do.

Is there a better way to check that all times are correct than look at  the clock for twelve hours?
        

self.hours= [list(range(49,54)), \
            list(range(57,60)), \
            list(range(55,59)), \
            list(range(67,71)), \
            list(range(84,88)), \
            list(range(73,77)), \
            list(range(100,105)), \
            list(range(60,66)), \
            list(range(89,93)), \
            list(range(80,84)), \
            list(range(93,97)), \
            list(range(77,80)), \
            list(range(49,54))]
        self.full_hour= list(range(107,110)) a

phenze, 2021-12-16:
You can use the simulator and set a time manually.
But i think that will be a little bit to hard to explain.

You can send be a full list of possible time values an i will check it for you.
In the meantime there is no better way than looking at it for twelve hours :D 

All is need is what should be displayed in every 5 minute step.
And a list of all possible hour values.

poa0314, 2021-12-16:
Hope that this is what you are looking for. I have typed it in next to the german equals.

I have seen that hour "TOLV" is missing the "T" and that pixel 60 is on at some times and should never be on.

[Urtavla.pdf](https://github.com/bk1285/rpi_wordclock/files/7727410/Urtavla.pdf)

  This class returns a given time as words (string)::
    """

    def __init__(self):
        self.prefix = "ES IST "     KLOCKAN ÄR
        self.minutes = ["",
            "FUNF NACH ", \         FEM ÖVER
            "ZEHN NACH ", \         TIO ÖVER
            "VIERTEL NACH ", \      KVART ÖVER
            "ZWANZIG NACH ", \      TJUGO ÖVER
            "FUNF VOR HALB ", \     FEM I HALV
            "HALB ", \              HALV
            "FUNF NACH HALB ", \    FEM ÖVER HALV
            "ZWANZIG VOR ", \       TJUGO I
            "VIERTEL VOR ", \       KVART I
            "ZEHN VOR ", \          TIO I
            "FUNF VOR "]            FEM I
        self.hours = ["ZWOLF", \    TOLV
            "EIN", \                ETT
            "ZWEI", \               TVÅ
            "DREI", \               TRE
            "VIER", \               FYRA
            "FUNF", \               FEM
            "SECHS", \              SEX
            "SIEBEN", \             SJU
            "ACHT", \               ÅTTA
            "NEUN", \               NIO
            "ZEHN", \               TIO
            "ELF", \                ELVA
            "ZWOLF"]                TOLV
        self.full_hour_suffix = " UHR" NOT USED

poa0314, 2021-12-16:
I found some cosmetic changes that place the "I" closer to the different minutes.
I include the modified file.
I guess that the original problems remains and appers round tvelve. I will look att that round midnight or lunchtime.

[time_swedish.txt](https://github.com/bk1285/rpi_wordclock/files/7729963/time_swedish.txt)

poa0314, 2021-12-17:
I think that I have solved the problems with "time_swedish", but it is difficult to check everything without spending a whole day in front of the clock.

When I am now almost satisfied how the clock works I am looking for a safe way to copy/clone the SD-card. I have heard that som of these tools are dangerous to use. Do you know anything regarding EaseUS or any other tool?

Uppdaterad fil.
[time_swedish.py.txt](https://github.com/bk1285/rpi_wordclock/files/7735442/time_swedish.py.txt)

FrankX0, 2021-12-17:
Nice to know the clock also speaks Swedish now!
Regarding backup and restore of your SD card: I would suggest [Etcher](https://www.balena.io/etcher/).

poa0314, 2021-12-18:
Is it possible to clone an SD-card including OS with Etcher?

I tried using Copy using Raspberry Pi. It works but without OS as I understand it.

I use Raspberry Pi Imager to install OS and language and wifi. It has an option #custom.img", is this a way to create a clone?

FrankX0, 2021-12-18:
With Etcher you clone the entire SD card.
As far as I know PI imager is to install an online image or a custom image you downloaded yourself.

poa0314, 2021-12-18:
Etcher was the perfect tool for me and worked without any problems.

I now have to finalize the mechanical build with a rusty steel frontplate.

The only thing I miss is a light sensor controlling the LEDs. Is this an ongoing job?

**I am very greatful for all excellent support!**

phenze, 2021-12-20:
Sorry i don't had a minute this weekend.

I have checked your changes. Everything looks good now 👍  Well done. I will ask Bernd to merge it back in the development branch.

I have a working light sensor. Needed code is already implemented.
All you have to do is to install a lux sensor (i have used tsl2561).

Then you can enable using the brightness sensor in the config file (`use_brightness_sensor = true`)

For more inspiration :
https://github.com/bk1285/rpi_wordclock/issues/139
https://github.com/bk1285/rpi_wordclock/issues/27

poa0314, 2021-12-20:
Excellent, du you also have info how to wire it?

phenze, 2021-12-21:
[image]

poa0314, 2022-01-21:
I have finally got a 2561 board and have connected it.
The type is: CJMCU-2561 Digital light sensor v0.9b.
I have connected it as described above. Only difference is that tha supply voltage was marked 5V on th board so iI connected it to 5V. SDA to pin 3 and SCL to pin 5.

I changed in the config file to.
# Set the brightness of the display (between 1 and 255)
brightness = 200
use_brightness_sensor = True

pi@raspberrypi:~/rpi_wordclock_phenze $ sudo python3 wordclock.py
2022-01-21 17:33:16 raspberrypi root[1214] INFO Software version: v1.0.0-dev-114-g052b87f
2022-01-21 17:33:16 raspberrypi root[1214] INFO Parsing /home/pi/rpi_wordclock_phenze/wordclock_config/wordclock_config.cfg
2022-01-21 17:33:16 raspberrypi root[1214] INFO Interface type set to gpio_low
2022-01-21 17:33:16 raspberrypi root[1214] INFO Wiring configuration
2022-01-21 17:33:16 raspberrypi root[1214] INFO   WCA_WIDTH: 11
2022-01-21 17:33:16 raspberrypi root[1214] INFO   WCA_HEIGHT: 10
2022-01-21 17:33:16 raspberrypi root[1214] INFO   Num of LEDs: 114
2022-01-21 17:33:16 raspberrypi root[1214] INFO   Wiring layout: bernds_wiring
2022-01-21 17:33:16 raspberrypi root[1214] INFO Setting language to swedish.
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 0: "feed_parser".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2022-01-21 17:33:16 raspberrypi root[1214] WARNING Defaulting to reference value for [plugin_test_strip] activate
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 1: "test_strip".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 2: "matrix".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 3: "restart".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 4: "leds_off".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 5: "time_in_seconds".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2022-01-21 17:33:16 raspberrypi root[1214] INFO Skipping plugin weather_forecast since it is set to activate=false in the config-file.
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 6: "tetris".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 7: "shutdown".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 8: "rainbow".
Using brigtness sensor : True
Importing sensor Library 
No Hardware I2C on (scl,sda)=(3, 2)
Valid I2C ports: ((1, 3, 2), (0, 1, 0))
2022-01-21 17:33:16 raspberrypi root[1214] WARNING Failed to import plugin time_default!
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 82, in __init__
    self.plugins.append(import_module('wordclock_plugins.' + plugin + '.plugin').plugin(self.config))
ValueError: No Hardware I2C on (scl,sda)=(3, 2)
Valid I2C ports: ((1, 3, 2), (0, 1, 0))
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 9: "sunrise".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 10: "snake".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 11: "ip_address".
2022-01-21 17:33:16 raspberrypi root[1214] INFO Imported plugin 12: "time_matrix".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2022-01-21 17:33:16 raspberrypi werkzeug[1214] INFO  * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 204, in <module>
    word_clock.run_forever()
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 189, in run_forever
    self.run()
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 156, in run
    self.plugin_index = self.default_plugin
AttributeError: 'wordclock' object has no attribute 'default_plugin'

What could be the problem?

FrankX0, 2022-01-21:
There is an error saying there is no hardware I2C.
You should enable this with ```
sudo raspi-config
```

poa0314, 2022-01-21:
I get a little bit further but still no clock running.
'pi@raspberrypi:~/rpi_wordclock_phenze $ sudo python3 wordclock.py
2022-01-21 21:14:43 raspberrypi root[1209] INFO Software version: v1.0.0-dev-114-g052b87f
2022-01-21 21:14:43 raspberrypi root[1209] INFO Parsing /home/pi/rpi_wordclock_phenze/wordclock_config/wordclock_config.cfg
2022-01-21 21:14:43 raspberrypi root[1209] INFO Interface type set to gpio_low
2022-01-21 21:14:43 raspberrypi root[1209] INFO Wiring configuration
2022-01-21 21:14:43 raspberrypi root[1209] INFO   WCA_WIDTH: 11
2022-01-21 21:14:43 raspberrypi root[1209] INFO   WCA_HEIGHT: 10
2022-01-21 21:14:43 raspberrypi root[1209] INFO   Num of LEDs: 114
2022-01-21 21:14:43 raspberrypi root[1209] INFO   Wiring layout: bernds_wiring
2022-01-21 21:14:43 raspberrypi root[1209] INFO Setting language to swedish.
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 0: "feed_parser".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2022-01-21 21:14:43 raspberrypi root[1209] WARNING Defaulting to reference value for [plugin_test_strip] activate
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 1: "test_strip".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 2: "matrix".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 3: "restart".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 4: "leds_off".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 5: "time_in_seconds".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2022-01-21 21:14:43 raspberrypi root[1209] INFO Skipping plugin weather_forecast since it is set to activate=false in the config-file.
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 6: "tetris".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 7: "shutdown".
2022-01-21 21:14:43 raspberrypi root[1209] INFO Imported plugin 8: "rainbow".
**Using brigtness sensor : True
Importing sensor Library 
No I2C device at address: 0x39**
2022-01-21 21:14:43 raspberrypi root[1209] WARNING Failed to import plugin time_default!
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 154, in __probe_for_device
    self.i2c.writeto(self.device_address, b"")
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/adafruit_bus_device/i2c_device.py", line 160, in __probe_for_device
    self.i2c.readfrom_into(self.device_address, result)
OSError: [Errno 121] Remote I/O error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 82, in __init__
    self.plugins.append(import_module('wordclock_plugins.' + plugin + '.plugin').plugin(self.config))
ValueError: No I2C device at address: 0x39
2022-01-21 21:14:44 raspberrypi root[1209] INFO Imported plugin 9: "sunrise".
2022-01-21 21:14:44 raspberrypi root[1209] INFO Imported plugin 10: "snake".
2022-01-21 21:14:44 raspberrypi root[1209] INFO Imported plugin 11: "ip_address".
2022-01-21 21:14:44 raspberrypi root[1209] INFO Imported plugin 12: "time_matrix".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2022-01-21 21:14:44 raspberrypi werkzeug[1209] INFO  * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 204, in <module>
    word_clock.run_forever()
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 189, in run_forever
    self.run()
  File "/home/pi/rpi_wordclock_phenze/wordclock.py", line 156, in run
    self.plugin_index = self.default_plugin
AttributeError: 'wordclock' object has no attribute 'default_plugin'

FrankX0, 2022-01-21:
It cannot find the sensor (at address 0x39).
Maybe check your wiring?

FrankX0, 2022-01-22:
Sensor should work.
Maybe first try to debug without the clock running.
Maybe this helps: https://www.instructables.com/Lux-Sensor-TSL2561-Raspberry-Pi/

poa0314, 2022-01-22:
My board do not have an ADDR pin. It has 4 pins 0V, 5V, SCL, GND.
I have tried to understand the information in your link above but this seems to be above my competence to understand.
during installation of the clock I have followed : https://github.com/phenze/rpi_wordclock/tree/master-python3
I the Readmefile it states:
⏳ In progress
Integration of a brightness sensor

Do this mean that code is missing to ues teh TSL2561 sensor?

FrankX0, 2022-01-22:
The code you are using does support the sensor.
The sensor itself is not detected in your case (`No I2C device at address: 0x39`).
Maybe it is at a different address.
Can you change following line: `address = 0x39 ## Device address` (at [this](https://github.com/phenze/rpi_wordclock/blob/de05cd2378ab7915eae41730d0f4b3b9293d36b8/wordclock_plugins/time_default/plugin.py#L120) location) to `0x29` and to `0x49`?
These are the other addresses supported by this sensor.

poa0314, 2022-01-22:
Is it in the file **plugin.py** that I should make the changes?

The file looks different from the file you have linked to and have no information about address.

**Copy of the rows regarding brightness sensor in my plugin.py:**

      self.brightness_mode_pos = config.getint('wordclock_display', 'brightness')
        self.brightness_change = 8

        self.use_brightness_sensor = config.getboolean('wordclock_display', 'use_brightness_sensor')

        print(('Using brigtness sensor : ' + str(self.use_brightness_sensor)))
        if self.use_brightness_sensor:
            print('Importing sensor Library ')
            import board
            import busio
            import adafruit_tsl2561
            i2c = busio.I2C(board.SCL, board.SDA)
            self.sensor = adafruit_tsl2561.TSL2561(i2c)
        # save current brightness for switching back from sleep mode
        self.wake_brightness = self.brightness_mode_pos

FrankX0, 2022-01-22:
You are apparently using a different branch than I was expecting.
For this implementation I could not find (yet) how (or if) to set the I2C address.

So, can you try the following:

- `sudo apt-get install i2c-tools`
- `i2cdetect -y 1`

This will list all I2C devices found.

poa0314, 2022-01-22:
Thank you for trying to help me to solv the problem!

This was the result.

pi@raspberrypi:~ $ sudo apt-get install i2c-tools
Läser paketlistor… Färdig
Bygger beroendeträd… Färdig
Läser tillståndsinformation… Färdig
i2c-tools is already the newest version (4.2-1+b1).
i2c-tools är satt till manuellt installerad.
Följande paket har installerats automatiskt och är inte längre nödvändigt:
  libfuse2
Använd ”sudo apt autoremove” för att ta bort det.
0 att uppgradera, 0 att nyinstallera, 0 att ta bort och 119 att inte uppgradera.
pi@raspberrypi:~ $ i2 detect -y 1
bash: i2: kommandot finns inte
pi@raspberrypi:~ $ i2detect -y 1
bash: i2detect: kommandot finns inte
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $

FrankX0, 2022-01-22:
That is a step forward!
So the sensor is at 0x29, not at 0x39.
No we need to figure out how to set the address in the Python code.

FrankX0, 2022-01-22:
Can you try to change
`self.sensor = adafruit_tsl2561.TSL2561(i2c)`
to
`self.sensor = adafruit_tsl2561.TSL2561(i2c,0x29)`

poa0314, 2022-01-23:
**Thank you very much!**
The light sensor is now working.
I am very impressed by the support I have got trying to build a Swedish Wordclock.
I am now only missing the frontplate in rusty steel but I am more familiar with that job than the programming.

Thank you once again for excellent support!

poa0314, 2022-02-02:
I am sorry to disturb you again!

Everything worked fine so I did clone the SD card to have a backup. After the clone nothing worked as it should not even with the original SD card. I think I did it correctly using clone in balena Etcher. It looked as if I used the wrong SD as source since the last updates on where not done. I have now made the last updates again but nothing works, no LEDs are working.
I tried to understad the logfile but no success.

I get Low voltage warnings could that be the problem.
I supply the Pi and the LEDs with 5V 10A powersupply. I do the power supply to the Pi via pin 2 not using the mini usb port.
I have had the same wiring all the time with no problems.

pi@raspberrypi:~ $ sudo python3 rpi_wordclock_phenze/wordclock.py
2022-02-02 10:26:57 raspberrypi root[1551] INFO Software version: v1.0.0-dev-114-g052b87f
2022-02-02 10:26:57 raspberrypi root[1551] INFO Parsing /home/pi/rpi_wordclock_phenze/wordclock_config/wordclock_config.cfg
2022-02-02 10:26:57 raspberrypi root[1551] INFO Interface type set to gpio_low
2022-02-02 10:26:57 raspberrypi root[1551] INFO Wiring configuration
2022-02-02 10:26:57 raspberrypi root[1551] INFO   WCA_WIDTH: 11
2022-02-02 10:26:57 raspberrypi root[1551] INFO   WCA_HEIGHT: 10
2022-02-02 10:26:57 raspberrypi root[1551] INFO   Num of LEDs: 114
2022-02-02 10:26:57 raspberrypi root[1551] INFO   Wiring layout: bernds_wiring
2022-02-02 10:26:57 raspberrypi root[1551] INFO Setting language to swedish.
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 0: "feed_parser".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2022-02-02 10:26:58 raspberrypi root[1551] WARNING Defaulting to reference value for [plugin_test_strip] activate
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 1: "test_strip".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 2: "matrix".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 3: "restart".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 4: "leds_off".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 5: "time_in_seconds".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2022-02-02 10:26:58 raspberrypi root[1551] INFO Skipping plugin weather_forecast since it is set to activate=false in the config-file.
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 6: "tetris".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 7: "shutdown".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 8: "rainbow".
Using brigtness sensor : True
Importing sensor Library 
2022-02-02 10:26:58 raspberrypi root[1551] INFO Selected "time_default" as default plugin
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 9: "time_default".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 10: "sunrise".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 11: "snake".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 12: "ip_address".
2022-02-02 10:26:58 raspberrypi root[1551] INFO Imported plugin 13: "time_matrix".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Exception ignored in thread started by: <bound method web_interface.threaded_app of <wordclock_interfaces.web_interface.web_interface object at 0x753bf778>>
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock_phenze/wordclock_interfaces/web_interface.py", line 71, in threaded_app
    self.app.run(host='0.0.0.0', port=port)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 990, in run
    run_simple(host, port, self, **options)
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 1052, in run_simple
    inner()
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 996, in inner
    srv = make_server(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 847, in make_server
    return ThreadedWSGIServer(
  File "/usr/lib/python3/dist-packages/werkzeug/serving.py", line 740, in __init__
    HTTPServer.__init__(self, server_address, handler)
  File "/usr/lib/python3.9/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.9/http/server.py", line 138, in server_bind
    socketserver.TCPServer.server_bind(self)
  File "/usr/lib/python3.9/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 98] Address already in use
2022-02-02 10:27:08 raspberrypi root[1551] INFO Running plugin time_default.
^C2022-02-02 10:29:50 raspberrypi root[1551] ERROR Error in plugin time_default.
2022-02-02 10:29:50 raspberrypi root[1551] ERROR PLEASE PROVIDE THE CURRENT SOFTWARE VERSION (GIT HASH), WHEN REPORTING THIS ERROR: v1.0.0-dev-114-g052b87f
Traceback (most recent call last):
[... 6 lines cut ...]
  File "/home/pi/rpi_wordclock_phenze/wordclock_interfaces/event_handler.py", line 81, in __wait_for
    self.condition.wait(waittime)
  File "/usr/lib/python3.9/threading.py", line 316, in wait
    gotit = waiter.acquire(True, timeout)
KeyboardInterrupt
Traceback (most recent call last):
[... cut ...]
  File "/home/pi/rpi_wordclock_phenze/wordclock_plugins/time_default/plugin.py", line 195, in run
    event = wci.waitForEvent(2)
  File "/home/pi/rpi_wordclock_phenze/wordclock_interfaces/event_handler.py", line 38, in waitForEvent
    self.__wait_for(lambda: self.event != self.EVENT_INVALID, seconds)
  File "/home/pi/rpi_wordclock_phenze/wordclock_interfaces/event_handler.py", line 81, in __wait_for
    self.condition.wait(waittime)
  File "/usr/lib/python3.9/threading.py", line 316, in wait
    gotit = waiter.acquire(True, timeout)
KeyboardInterrupt

pi@raspberrypi:~ $

phenze, 2022-02-02:
Can you please open a new issue ?
Then we can discuss your problems there.

phenze, 2022-02-02:
The error message is `Address already in use`.
So the wordclock software cannot start the local Web server.

Seems that there is some other process running on port 80. Did you install anything else ?

Perhaps another instance of the software is running?
You can check this with:
`ps aux | grep python`

poa0314, 2022-02-02:
I removed Crontab and started it manually.

Then I did not get this error message but still no LEDs.

pi@raspberrypi:~ $ sudo python3 rpi_wordclock_phenze/wordclock.py
2022-02-02 17:18:14 raspberrypi root[2276] INFO Software version: v1.0.0-dev-114-g052b87f
2022-02-02 17:18:14 raspberrypi root[2276] INFO Parsing /home/pi/rpi_wordclock_phenze/wordclock_config/wordclock_config.cfg
2022-02-02 17:18:14 raspberrypi root[2276] INFO Interface type set to gpio_low
2022-02-02 17:18:14 raspberrypi root[2276] INFO Wiring configuration
2022-02-02 17:18:14 raspberrypi root[2276] INFO   WCA_WIDTH: 11
2022-02-02 17:18:14 raspberrypi root[2276] INFO   WCA_HEIGHT: 10
2022-02-02 17:18:14 raspberrypi root[2276] INFO   Num of LEDs: 114
2022-02-02 17:18:14 raspberrypi root[2276] INFO   Wiring layout: bernds_wiring
2022-02-02 17:18:14 raspberrypi root[2276] INFO Setting language to swedish.
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 0: "feed_parser".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2022-02-02 17:18:15 raspberrypi root[2276] WARNING Defaulting to reference value for [plugin_test_strip] activate
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 1: "test_strip".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 2: "matrix".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 3: "restart".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 4: "leds_off".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 5: "time_in_seconds".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2022-02-02 17:18:15 raspberrypi root[2276] INFO Skipping plugin weather_forecast since it is set to activate=false in the config-file.
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 6: "tetris".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 7: "shutdown".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 8: "rainbow".
Using brigtness sensor : True
Importing sensor Library 
2022-02-02 17:18:15 raspberrypi root[2276] INFO Selected "time_default" as default plugin
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 9: "time_default".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 10: "sunrise".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 11: "snake".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 12: "ip_address".
2022-02-02 17:18:15 raspberrypi root[2276] INFO Imported plugin 13: "time_matrix".
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2022-02-02 17:18:15 raspberrypi werkzeug[2276] INFO  * Running on http://0.0.0.0:80/ (Press CTRL+C to quit)
2022-02-02 17:18:26 raspberrypi root[2276] INFO Running plugin time_default.

poa0314, 2022-02-02:
Problem solved!
It was a bad soldering that caused the problem.

phenze, 2022-02-03:
Unfortunately that is most of the time the problem.