bk1285/rpi_wordclock #215: Nearly finished..

tom4727, 2021-12-06:
Moin. I just see the finishline :-) 

Just after 

cd ~/rpi_wordclock
sudo python wordclock.py

Raspberry says:

ModuleNotFoundError: No module named „ConfigParser“.

Can anybody help?

Calimerorulez, 2021-12-06:
https://techglimpse.com/fix-no-module-named-import-error/

Maybe this helps?

phenze, 2021-12-07:
Which python version are you using ?
```python --version```

Which Raspian Version have you installed ?

You should use python3 instead of python2.

Take a look at this discussion:
https://github.com/bk1285/rpi_wordclock/issues/211

tom4727, 2021-12-11:
Moin, my version is Python 3.9.2. another prob is that sudo scons is allready missing.

tom4727, 2021-12-11:
> 
$pip install configparser
Collecting configparser
Using cached configparser-3.3.0r2.tar.gz
Building wheels for collected packages: configparser
Running setup.py bdist_wheel for configparser
Stored in directory: /root/.cache/pip/wheels/1a/3e/f9/d34006ad6b1edfe5006aa704f5ee305c553344a7a6d8550c29
Successfully built configparser
Installing collected packages: configparser
Successfully installed configparser-3.3.0.post2
install: Fehlender Zieldatei‐Operand hinter 'configparser'
„install --help“ liefert weitere Informationen.
bash: Collecting: Kommando nicht gefunden.
bash: Using: Kommando nicht gefunden.
bash: Building: Kommando nicht gefunden.
bash: Running: Kommando nicht gefunden.
bash: Stored: Kommando nicht gefunden.
bash: Successfully: Kommando nicht gefunden.
bash: Installing: Kommando nicht gefunden.

tom4727, 2021-12-11:
sudo: scons: Befehl nicht gefunden

tom4727, 2021-12-11:
thank you for your help and kindness. now there is that issue:

sudo pip3 install ConfigParser
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ConfigParser
  Downloading https://www.piwheels.org/simple/configparser/configparser-5.2.0-py3-none-any.whl (19 kB)
Installing collected packages: ConfigParser
Successfully installed ConfigParser-5.2.0
pi@raspberrypi:~ $ cd ~/rpi_wordclock
sudo python wordclock.py
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 1, in <module>
    import ConfigParser
ModuleNotFoundError: No module named 'ConfigParser'

FrankX0, 2021-12-11:
I would advise to use the develop branch:
`https://github.com/bk1285/rpi_wordclock/tree/develop`
with its installation instructions.

tom4727, 2021-12-12:
thx FrankX0 :-)  i think this solved my issue. the cronjob is now working (i think). now i will get all the components together and will have a look, if all is working togehter.

RahulYadav2207, 2022-01-31:
i also got same error

RahulYadav2207, 2022-01-31:
@tom4727 what to do if i got the same error which you got No module named 'ConfigParser'

phenze, 2022-01-31:
Reading the instructions here :

* https://github.com/bk1285/rpi_wordclock/tree/develop
* https://github.com/bk1285/rpi_wordclock/issues/211

or all the instructions in this issue.

I think your main problem is that you dont use python3 (https://github.com/bk1285/rpi_wordclock/issues/211#issuecomment-966865031)

RahulYadav2207, 2022-01-31:
i used

RahulYadav2207, 2022-01-31:
@phenze can you please explain me in detail how to setup all things plz

phenze, 2022-01-31:
Did you switch to the development branch ?

`git checkout --track origin/develop`

RahulYadav2207, 2022-01-31:
What's this how can i access this in rpi terminal

phenze, 2022-02-01:
That is all explained here:
https://github.com/bk1285/rpi_wordclock/tree/develop

```
git clone https://github.com/bk1285/rpi_wordclock.git
cd rpi_wordclock/
git checkout --track origin/develop
```

Do you know how to use a Linux Terminal and what the commands are doing ?
If not do this tutorial:
* https://ubuntu.com/tutorials/command-line-for-beginners#1-overview

For a basic overview what git does:
* https://rogerdudler.github.io/git-guide/

FrankX0, 2022-02-01:
@bk1285: this might be a good time to make the develop branch the new main branch?

bk1285, 2022-09-01:
Done, @FrankX0 :)

closed 2022-09-01 by bk1285