bk1285/rpi_wordclock #218: Kein Zugriff über den Browser / No access via browser

echo1483, 2022-02-08:
Hallo,  meine Wordclock läuft schon lange ohne Probleme. Seit gestern kann ich nicht mehr über den Browser auf die Uhr zugreifen. Wer weis Rat. Danke

Hello, my word clock has been running without problems for a long time. Since yesterday I can no longer access the watch via the browser. Who knows advice. thank you

Fehler / error 

Control Colors fa fa-question-circle The Raspberry Wordclock fa fa-link [Project page](https://github.com/bk1285/rpi_wordclock)
fa fa-users [Contributors](https://github.com/bk1285/rpi_wordclock/graphs/contributors)
fa fa-cogs [API of this clock](http://192.168.178.40:8080/api#/API)
 fa fa-caret-left fa fa-circle fa fa-caret-right 

[image]

phenze, 2022-02-08:
That is already fixed in the development branch.
https://github.com/bk1285/rpi_wordclock/issues/177

This Branch needs python3. So i think you need to update the whole installation or merge this changes manually.

echo1483, 2022-02-08:
Reicht es wenn ich das rpi_wordclock Verzeichnis neu kopiere. Danke für die Antwort.

Is it enough if I copy the rpi_wordclock directory again. Thanks for the answer.

phenze, 2022-02-08:
Entweder du kopierst dir die Änderungen aus folgendem Commit manuell in das Verzeichnis:

https://github.com/FrankX0/rpi_wordclock/commit/0edce92307fefca6f44287f07e958324d2fa055c

Oder du benutzt den development Brand auf Python3 Basis. Was hier genau zu tun ist ist hier genau beschrieben:
https://github.com/bk1285/rpi_wordclock/tree/develop

Im Grunde ist damit vieles einfacher und du bekommst die neuesten Updates. Der Mehraufwand lohnt sich also.

phenze, 2022-02-08:
Bist du auf dem richtigen Branch unterwegs ?
Hast du den Befehl ausgeführt :
`git checkout --track origin/develop`

Ansonsten bitte mal im Verzeichnis `/home/pi/rpi_wordclock/`folgende Befehle eingeben:

`git status` und  `git remote -v` und mir hier die Ausgabe reinschreiben.

Außerdem noch `python3 --version`

echo1483, 2022-02-08:
Danke für deine Hilfe, ich selbst habe keinen Schimmer in Richtung Raspberry.
Hier die Befehle, sagen mir nichts.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Feb  8 19:24:06 2022 from 192.168.178.29
pi@raspberrypi:~ $ git checkout --track origin/develop
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
pi@raspberrypi:~ $ cd ~/rpi_wordclock
pi@raspberrypi:~/rpi_wordclock $ git status
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
pi@raspberrypi:~/rpi_wordclock $  git remote -v
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
pi@raspberrypi:~/rpi_wordclock $ python3 --version
Python 3.5.3
pi@raspberrypi:~/rpi_wordclock $

echo1483, 2022-02-08:
Ich habe alle Befehle wie dort beschrieben eingegeben.
Ab hier bekomme ich Probleme:

pi@raspberrypi:~/rpi_wordclock $ git clone https://github.com/bk1285/rpi_wordclock.git
Klone nach 'rpi_wordclock' ...
remote: Enumerating objects: 4788, done.
remote: Counting objects: 100% (1823/1823), done.
remote: Compressing objects: 100% (1523/1523), done.
remote: Total 4788 (delta 433), reused 1640 (delta 294), pack-reused 2965
Empfange Objekte: 100% (4788/4788), 30.00 MiB | 1.17 MiB/s, Fertig.
Löse Unterschiede auf: 100% (2292/2292), Fertig.
Checke Dateien aus: 100% (245/245), Fertig.

Nu kommt die Fehlermeldung.

pi@raspberrypi:~/rpi_wordclock $ git checkout --track origin/develop
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git

Liegt es an meine Raspian Version?

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

echo1483, 2022-02-08:
Macht es Sinn alles zu löschen und eine neu Raspbian-Version aufzuspielen. 
Und welche Version sollte ich dann nehmen. Mein Raspberry Pi Zero W ist einer der ersten Generation.

phenze, 2022-02-08:
Okay dann bitte folgendes ausführen (nacheinander):
```
cd ~
sudo rm -Rf rpi_wordclock
git clone https://github.com/bk1285/rpi_wordclock.git
cd rpi_wordclock
git checkout --track origin/develop
```

Dann sollte es klappen mit `python3 wordclock.py`

[image]

phenze, 2022-02-08:
Wenn das alle nicht klappt macht eine neue Raspbian Version Sinn ja.

Ich bin auf Buster unterwegs

echo1483, 2022-02-08:
Haut nicht hin wieder Fehler.

pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ sudo rm -Rf rpi_wordclock
pi@raspberrypi:~ $ git clone https://github.com/bk1285/rpi_wordclock.git
Klone nach 'rpi_wordclock' ...
remote: Enumerating objects: 4788, done.
remote: Counting objects: 100% (1823/1823), done.
remote: Compressing objects: 100% (1523/1523), done.
remote: Total 4788 (delta 433), reused 1640 (delta 294), pack-reused 2965
Empfange Objekte: 100% (4788/4788), 30.00 MiB | 1.34 MiB/s, Fertig.
Löse Unterschiede auf: 100% (2292/2292), Fertig.
Checke Dateien aus: 100% (245/245), Fertig.
pi@raspberrypi:~ $ cd rpi_wordclock
pi@raspberrypi:~/rpi_wordclock $ git checkout --track origin/develop
Branch develop konfiguriert zum Folgen von Remote-Branch develop von origin.
Zu neuem Branch 'develop' gewechselt

pi@raspberrypi:~/rpi_wordclock $ python3 wordclock.py
Traceback (most recent call last):
  File "wordclock.py", line 17, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 4, in <module>
    from flask_restx import Api, Resource, fields
  File "/usr/local/lib/python3.5/dist-packages/flask_restx/__init__.py", line 4, in <module>
    from . import fields, reqparse, apidoc, inputs, cors
  File "/usr/local/lib/python3.5/dist-packages/flask_restx/reqparse.py", line 19, in <module>
    from .model import Model
  File "/usr/local/lib/python3.5/dist-packages/flask_restx/model.py", line 21, in <module>
    from jsonschema import Draft4Validator
  File "/usr/local/lib/python3.5/dist-packages/jsonschema/__init__.py", line 56
    raise AttributeError(f"module {__name__} has no attribute {name}")
                                                                    ^
SyntaxError: invalid syntax
pi@raspberrypi:~/rpi_wordclock $

Hast du Raspbian Buster Lite  drauf ?
Ich denke ich lösche alles und  baue alles neu auf. Schade meine Uhr läuft wenn ich mich nicht irre seit 2019 ohne große Probleme.

phenze, 2022-02-08:
Ja Buster Lite. kannst es aber auch mal mit dem neusten (bullseye)
probieren.

echo1483, 2022-02-09:
Sie läuft wieder mit Buster Lite gabs Probleme. Beim neuen Bullseye Lite lief die Installation wie von selbst. 
Also bei Probleme alle formatiere und nach dieser Anleitung vorgehen.
https://github.com/bk1285/rpi_wordclock/tree/develop

Vielen Dank phenze
echo aus NRW (Kempen)

phenze, 2022-02-10:
Super freut mich!