bk1285/rpi_wordclock #220: Bavarian / Swabian noch working

FelixNudrec, 2022-03-29:
Hej all,

I hant to have the East-German "Dreiviertel". The Swabian mode should support this. However, I can't get this to work. I changed the language to swabian or bavarian in the Wordclock_config.cfg. In both cases, the clock does not start to light up.

What do I have to do to get the "Dreiviertel" in?
In the end it would be good if the language query in the config file said what was available to choose from and how to write it.

THX for help

phenze, 2022-03-29:
The changes needed for this are not in the repo yet. You have to apply this pull request:

https://github.com/bk1285/rpi_wordclock/pull/212

Alternatively you can switch to my fork:
https://github.com/phenze/rpi_wordclock/tree/develop

OBetzi, 2022-03-29:
@phenze and @FelixNudrec :
I think the problem is caused by not listed SWABIAN and BAVARIAN language in wordclock_config.cfg language_options section. This was the problem to me...

FelixNudrec, 2022-03-29:
@phenze But at your fork I am not able to change my start colour right?
How can I apply this pull request? ( Sorry I'm not the best in the Git-Game)

@OBetzi Do you think I can write a new language in the options and then it will work?

OBetzi, 2022-03-29:
I think so.

OBetzi, 2022-03-29:
for example (not the right letters, but for functions this is not a problem): 
swabian=["DMZYTLISCHB",
    "PFÜFVIERTUX",
    "ZWÄNZGKZÄÄG",
    "VORABJHAUBI",
    "ZWÖISÄCHSIQ",
    "ACHTIMVOUFI",
    "FÜFIRSIBLIT",
    "ONZÄNIADRÜY",
    "VIERIFNÜNIB",
    "HEISUZWÖLFI"]

OBetzi, 2022-03-29:
For my case, I have also an other stencil (Swabian not equal to German). I also do not use the generated stencil layout --> therefor no need the use a "right" one in the config file.

phenze, 2022-03-29:
okay then feel free to create a pull request for that so that you dont have to do all the changes with every update ;).

OBetzi, 2022-03-29:
Software development and especially GIT is not my best skill - but of there is time in the future I will do it ;-)

FelixNudrec, 2022-03-30:
> @FelixNudrec In my fork you can use all the features from the original branch. Currently it is fully merged with the original one.
> 
> Please take care to use the develop branch with python3 support.
> 
 Thanks that worked.

closed 2022-03-30 by FelixNudrec