bk1285/rpi_wordclock #221: Front/Stencil Wordclock

Silence667, 2022-04-19:
Ich mache hier nochmal die Frage auf. Also ich habe jetzt die Clock soweit zusammengebaut und auch (bis auf die Schalter und den Lichtsensor (Amazon lässt auf sich warten) verbaut. Jetzt habe ich im Moment weiterhin das Problem, dass ich nicht genau weiß wie ich die Front konstruiere und in welchem Versand ich diese erstellen könnte. Ich habe schon gegoogelt, aber vielleicht bringen mich Erfahrungsberichte zur Front und kleine Tipps weiter.
Danke schonmal für die Hilfe und Respekt für den großartigen Code.

phenze, 2022-04-20:
Fangen wir doch mal von vorne an.
Magst du vielleicht mal ein paar Bilder deiner Uhr schicken damit man sich das besser vorstellen kann ?
Was genau ist dein Problem ? leuchten die LED's schon und nur falsch ?

Die Buttons brauchst du nicht unbedingt. Das kann man mittlerweile alles über die Software steuern.

Bitte beachte beim aufsetzen der Software dass du den python-3 Branch (= develop) und eine aktuelles Rasbian verwendest.

Silence667, 2022-04-21:
Also das Programm läuft ohne Probleme, aber ich weiß bisher noch nicht, wie ich den Helligkeitssensor konfiguriert bekomme. Ich habe zunächst das Gehäuse im 3d-Druck gedruckt und danach verschweißt. Daraufhin habe ich die LEDs verlegt und mit Heißkleber fixiert. Das ganze betreibe ich im Moment mit einem Raspberry pi zero 2 w, auf dem das neuste Bullseye lite läuft. Die Software wurde mit dem TUT von dieser Seite durchgeführt und der cronjob in die Bootsequenz eingefügt. Jetzt hänge ich noch an der Front, die ich gerne aus Alu oder aus Edelstahl hätte, damit ich davor ein dünnes Holz legen kann (somit sieht man im ausgeschalteten Zustand nicht das Endprodukt). Dann wollte ich gerne den Helligkeitssensor konfigurieren, aber ich weiß nicht wie. Danke für Deine Hilfe und liebe Grüße. 

[image]
[image]
[image]

[image]

phenze, 2022-04-22:
Schau mal hier unter "Brightness Sensor". Da habe ich das erklärt:
https://github.com/phenze/rpi_wordclock/tree/develop

Deiner sollte derselbe sein den auch ich verwende.
Du musst dann nur auf meinen Fork (branch develop) wechseln.

Falls du das selber implementieren willst findest du den code dazu hier:
https://github.com/phenze/rpi_wordclock/blob/c994471b063b1332601783779307fed890e094bf/wordclock_plugins/time_default/plugin.py#L123

https://github.com/phenze/rpi_wordclock/blob/c994471b063b1332601783779307fed890e094bf/wordclock_plugins/time_default/plugin.py#L152

Zur Frontplatte habe ich dxf Dateien (45x45):
https://www.dropbox.com/sh/dwc92gzmupnocu4/AAD8YmHkXtBtVgNJ4zs4-RJ2a?dl=0

Eventuell kannst du damit ja was anfangen.

Deine Uhr sieht super aus! Tolle Arbeit

Silence667, 2022-04-22:
Danke erstmal für die schnelle Antwort 😀.

Ich merke aber, das ich ein step by step Tutorial benötige, weil ich doch nicht ganz in der Materie drin bin. 
Wie gehe ich bei dem Unterpunkt:
"
Set use_brightness_sensor config value to true

# Set the brightness of the display (between 1 and 255)
brightness = 200
use_brightness_sensor = True
" vor?

Gibt es Erfahrungen zu Fräsanbietern der DXF Datei?

Danke für die bisherige Hilfe.

phenze, 2022-04-24:
Das musst du in deiner Config Datei unter wordclock_config/wordclock_config.cfg einstellen.

Näheres ist hier erklärt:
https://rpi-wordclock.readthedocs.io/en/master/doc_software_setup.html#the-wordclock-software

Zwecks Fräsanbieter habe ich leider keine Erfahrung.

jimniki, 2022-10-13:
Any chance of telling us or showing us how to wire up the brightness sensor? Cheers. Jim.