bk1285/rpi_wordclock #228: color changing time

jimniki, 2022-10-11:
Hello all. I'm making my third clock and I'd like to have no buttons. With the web connect happening, I'm happy to adjust from there if need be. What I would love is to be able to set the time (including the words "IT IS" if possible) to use a slow rainbow effect. I can do this with buttons. Is there a simple solution to set this as a default on startup (without using any physical buttons)? Or via the web html? I'd like this because my last clock had very weak pixels for the "IT IS" wording as they were on 24/7 for several years. These pixels don't last forever and I was thinking that if I can rotate the rbg internal leds, (also at a selectable brightness) they may last longer.

jimniki, 2022-10-21:
.

closed 2022-10-21 by jimniki

FrankX0, 2022-10-21:
Hi @jimniki,
Does this mean you found/developed a solution?

jimniki, 2022-10-22:
Sadly, no. I just created the clock with buttons as per usual. I just press the right? button a couple of times to force the time into color changing mode and leave it at that. It's a pity I can't set this as a boot up default.

FrankX0, 2022-10-23:
I might be able to help. But please note that there only a few active developers. So please be patient.
I'll have a look, probably during coming week.

jimniki, 2022-10-23:
Ta Frank. Much appreciated. It might also solve another mystery. Since we
can't default the clock colours to the changing rainbow colors on bootup, I
manually set it with a couple of button presses. And you know what, hours
later it changes back to a fixed color fading pattern. It's done this on
all my clock builds? It doubt this is programmed, is it? Once again, do
whatever you can do at your own pace when you feel like it.  Cheers from
au.. Jim

FrankX0, 2022-10-25:
Hi Jim,
I just created a simple fix, by allowing the color cycling to be enabled from the config file.
Please see the pull request #231

jimniki, 2022-10-25:
Frank, you the man... very much appreciated...
Sincerely, Jim.

jimniki, 2022-10-30:
Hello again Frank. Just a quick one for you if and when you have a minute.
(please let me know if you don't want me emailing you again...no probs at
all.)

I know a year or 2 ago the brightness level sleep setting in the config
file wasn't working. Should it be working these days or is it still a
problem? The reason I ask is because with the color cycling you helped me
with, it isn't working. Could it be an issue with just the color cycling
mode?

Thanks again mate.
Jim (from Australia).

FrankX0, 2022-10-30:
No Problem. Although it would be better to open an issue if you want to report a defect.

Anyway, this is an actual defect. When in color cycling mode, the sleep setting is not taken into account.
I'll have a look how to fix this.

Kind regards,
Frank (from the Netherlands).

FrankX0, 2022-10-30:
Can you try the second [commit](https://github.com/bk1285/rpi_wordclock/pull/231/commits/ca29baa75c4de9a81a4af64efea86c57a9f5b169)?