bk1285/rpi_wordclock #232: Two plug in issues - MATRIX with Time and SECONDS

R0lleston, 2022-10-26:
We have two issues with plugins
1) MATRIX WITH TIME - slow and jerky display of the matrix green fonts - you can visually see it refreshing from left to right. The MATRIX without time works flawlessly. The Time variant used to work correctly in past builds

2) SECONDS - this has been reported previously - but the seconds skip at times - quite noticeable every 10 seconds - almost as if the plug in is trying to play catch up. I'm using a pi Zero so it could be a processing issue, but im not convinced it is

bk1285, 2022-11-01:
Thanks for reporting, @R0lleston. Could be related to the transitions, which are by now involved in the rendering of the current time... https://github.com/bk1285/rpi_wordclock/blob/ee7f2540fde915bda0ca666c7898a806384062cd/wordclock_tools/wordclock_display.py#L349

... I'll try to reproduce that on my clock.

FrankX0, 2023-03-08:
Hi Bernd, can you give this one priority?
I especially miss that I cannot show off with the Matrix with time plugin ;-)
Thanks in advance,
Frank.