bk1285/rpi_wordclock #234: Sleep brightness setting not supported in color cycling mode

FrankX0, 2022-10-30:
As reported by @jimniki

jimniki, 2022-10-31:
Hello Frank. I made the changes. No go. No lights on the clock at all. It was only when I hashed out all your +"additions" to the plugin.py file that it worked again. Maybe a typo somewhere? *shrugs*. Thanks heaps for your efforts btw...

FrankX0, 2022-10-31:
Hi Jim,
I cannot reproduce. Can you get the output from the terminal when you start the Python script?
This probably shows the error which causes no lights on your clock.
Thanks,
Frank.

jimniki, 2022-10-31:
It works on yours? That's excellent! It must be something I'm doing.... I cut/pasted your whole modified plugin.py file to avoid any typos on my end and with the flag in the config.cfg file set to True and False I still got no lights until I hashed out the modded lines...  I'll try again now... cheers.

jimniki, 2022-10-31:
Here's the errors when I start it up. No errors when I hash out the modded lines.

pi@wordclock1:~/rpi_wordclock $ sudo python3 wordclock.py
2022-11-01 08:37:22 wordclock1 root[1304] INFO Software version: v1.0.0-dev-138-g1b10921
2022-11-01 08:37:22 wordclock1 root[1304] INFO Parsing /home/pi/rpi_wordclock/wordclock_config/wordclock_config.cfg
2022-11-01 08:37:22 wordclock1 root[1304] INFO Interface type set to gpio_low
2022-11-01 08:37:22 wordclock1 root[1304] INFO Wiring configuration
2022-11-01 08:37:22 wordclock1 root[1304] INFO   WCA_WIDTH: 11
2022-11-01 08:37:22 wordclock1 root[1304] INFO   WCA_HEIGHT: 10
2022-11-01 08:37:22 wordclock1 root[1304] INFO   Num of LEDs: 114
2022-11-01 08:37:22 wordclock1 root[1304] INFO   Wiring layout: bernds_wiring
2022-11-01 08:37:22 wordclock1 root[1304] INFO Setting language to english.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin weather_forecast since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 0: "leds_off".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin feed_parser since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] WARNING Defaulting to reference value for [plugin_test_strip] activate
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 1: "test_strip".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin time_as_words_german since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin matrix since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin tetris since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 2: "ip_address".
unexpected indent (plugin.py, line 181)
2022-11-01 08:37:22 wordclock1 root[1304] WARNING Failed to import plugin time_default!
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 82, in __init__
    self.plugins.append(import_module('wordclock_plugins.' + plugin + '.plugin').plugin(self.config))
  File "/home/pi/rpi_wordclock/wordclock_plugins/time_default/plugin.py", line 181
    has_run = True                                                                               #
IndentationError: unexpected indent
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 3: "shutdown".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 4: "time_in_seconds".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin time_as_words_dutch since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin sunrise since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin snake since it is set to activate=false in the config-file.
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 5: "rainbow".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Imported plugin 6: "restart".
2022-11-01 08:37:22 wordclock1 root[1304] INFO Skipping plugin time_matrix since it is set to activate=false in the config-file.
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 204, in <module>
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
    word_clock.run_forever()
  File "/home/pi/rpi_wordclock/wordclock.py", line 189, in run_forever
    self.run()
  File "/home/pi/rpi_wordclock/wordclock.py", line 156, in run
    self.plugin_index = self.default_plugin
AttributeError: 'wordclock' object has no attribute 'default_plugin'
pi@wordclock1:~/rpi_wordclock $

FrankX0, 2022-11-01:
It is complaining about indents: 
```
File "/home/pi/rpi_wordclock/wordclock_plugins/time_default/plugin.py", line 181
has_run = True #
IndentationError: unexpected indent
```
It is complaining about row `has_run = True #` on line 181, but it is actually on line 151.
When doing a copy of code, check for space to tab conversions. And maybe remove the # at the end of the line.

jimniki, 2022-11-01:
will do. I'll let you know how I go.
cheers.

jimniki, 2022-11-01:
Hi Frank. I thought I fixed plugin.py file. These are the errors I am getting now. Sorry, I am a noob. Thoughts?

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 204, in <module>
 * Serving Flask app "wordclock_interfaces.web_interface" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
    word_clock.run_forever()
  File "/home/pi/rpi_wordclock/wordclock.py", line 189, in run_forever
   Use a production WSGI server instead.
 * Debug mode: off
    self.run()
  File "/home/pi/rpi_wordclock/wordclock.py", line 156, in run
    self.plugin_index = self.default_plugin
AttributeError: 'wordclock' object has no attribute 'default_plugin'

FrankX0, 2022-11-01:
The supplied information is as expected, except for `AttributeError: 'wordclock' object has no attribute 'default_plugin'`.
This means that there is still an error in the time_default plugin.
Can you get me the actual plugin.py file from time_default, so I can have a look what's wrong with it?
If you are not sure how, you could use e.g. WinSCP if you have SSH enabled.

jimniki, 2022-11-02:
Hello mate. I am going crazy. Everything looks to me. I'm sure it's
something silly only a noob can miss. This file should be identical to
yours...

closed 2023-03-07 by bk1285