bk1285/rpi_wordclock #235: Is the Raspberry pico w compatible with this wordclock?

jimniki, 2022-11-02:
Hello again. Did a search but found no references to "pico". Assume not. Hard to find any pi's for sale at a reasonable price. Anyone using an Orange pi?

FrankX0, 2022-11-02:
Hi Jim. The Pico W does not run Linux. So this implementation does not support the Pico.
My guess is you should be able to get this to run on other platforms running Linux.
But I guess there is only one way to find out ;-)

jimniki, 2022-11-02:
Any non rasp pi sbc anyone  can recommend to work?

maxce-ok, 2023-01-03:
Hi Jim, i just finished a first version of a running Wordclock on the Pico-W. It uses a time server to update a DS3231 Chip and displays very simple the Time as words on the same Layout like in this repository. Yes it is possible to use a pico-W. But actually only very limited functions.

bk1285, 2023-03-07:
@maxce-ok feel free to share any progress. Also a minimal version sounds promising... :)