bk1285/rpi_wordclock #237: Short question

basholtman, 2022-12-24:
In 2019 I set my color and brightness to a default value in the code not the config file. So far I found the brightness setting in wordclock_display.py

but I can’t find de color setting in the code? Does anyone know the specific file? 

Problem is that I build a wordclock of 1m by 1m an not all leds have the same color (white and some white/red. While in the strandtest all the less light up with the exact same color, so the hardware seems right.

FrankX0, 2022-12-27:
Not sure which branch you used, but my guess is:
```
        self.bg_color = wcc.BLACK  # default background color
        self.word_color = wcc.WWHITE  # default word color
        self.minute_color = wcc.WWHITE  # default minute color
```
in `wordclock_plugins/time_default/plugin.py`

basholtman, 2023-01-08:
trying a clean install now

closed 2023-01-08 by basholtman