bk1285/rpi_wordclock #238: Web Interface: Simple Example for an APP

PeterBrenden, 2023-01-07:
Hello to everybody,
first of all MANY THANKS to Bernd! Great project. Sofar I have extended it with weeksday, an alarm clock and a new language "allemanic". Everything is working fine, even the API, which I was able to extend, too.
My problem: I have no idea how to excess the API from an APP (to control the wordclock with a smart phone).
Is the anybody out there (sorry stolen from Pink Floyd), how maybe could show me some lines of code (like the *.html) , e.g. one button sending directly a "POST" or a "GET"?
Thanks a lot in advance and have a nice evening
Best regards
Peter

FrankX0, 2023-01-07:
Well, it already has a build-in "APP": the webinterface.
Just go to the IP address of your clock on your smart phone.
On an iPhone you can even then put it on your Home Screen: then it actually feels like a normal APP.

PeterBrenden, 2023-01-08:
Hi Frank,
thanks for the feedback. You are right, this part is working fine. 
Browser => http://192.168.2.139/api#/API => 
[image]
[image]
So e.g. 4. steps are required to change the brigthness.
1. "POST, 2."Try out", 3. Enter new value, 4. "Execute"

Means, I'm searching for the code, which I can adapt: change brigthness with one buttoen (taking the "Execute" button, move it to another site, change style and so on)

Does this descripton make sense to you, or is my writing still bad, not understandable (don't worry, just let me know :)
Many thanks
Peter

FrankX0, 2023-01-08:
Hi Peter. Just to be clear: you are aware that this APP implementation already exists?
In your case http://192.168.2.139
Regards,
Frank.

PeterBrenden, 2023-01-08:
Hi Frank,
sorry for beeing a pain...
I'm not that sure, maybe if have an old version?
With http://192.168.2.139 I'll get:
[image]
and then (pressing "API..."
[image]
Best regards
Peter

FrankX0, 2023-01-08:
Hi Peter. This is a known issue: #218 
Best would be to start using the latest (Python3) branch.
Please let me know if this is possible for you.

PeterBrenden, 2023-01-08:
Hi Frank,
I updated python to 3.11, but still having the same problem.
I tryed your "new" webinterface (https://github.com/FrankX0/rpi_wordclock/commit/0edce92307fefca6f44287f07e958324d2fa055c)?
Ending up with the error "module web could not be imported" and installing endet up with some more errors...
I'm so sorry...
Best regards
Peter

FrankX0, 2023-01-08:
Did you update Python or did you install the latest wordclock branch?
If not done so, you should follow the [Software setup](https://rpi-wordclock.readthedocs.io/en/main/doc_software_setup.html#software-setup).

PeterBrenden, 2023-01-08:
I think I give up.
after "sudo pip3 install -r requirements.txt", I'll get
 Could not find a version that satisfies the requirement astral==2.2 (from -r requirements.txt (line 2)) (from versions: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.6.1, 0.6.2, 0.7, 0.7.1, 0.7.2, 0.7.3, 0.7.4, 0.7.5, 0.8, 0.8.1, 0.8.2, 0.9, 1.0, 1.1, 1.2, 1.3, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 1.4, 1.5, 1.6, 1.6.1, 1.7, 1.7.1, 1.8, 1.9, 1.9.1, 1.9.2, 1.10, 1.10.1)
No matching distribution found for astral==2.2 (from -r requirements.txt (line 2))
And similar to allmost every next install
Best regards
Peter

FrankX0, 2023-01-09:
Hi Peter. Don't give up. We'll get this sorted out.
From the reports I conclude you are running an outdated Python3 installation.
Can you execute `python3 --version`?
Are you running the latest Pi OS version (bullseye)?

PeterBrenden, 2023-01-09:
Hi Frank,
thanks for your patience, I realy apprechiate that!
 
After struggeling with lots of error message, I thought nearly the same as 
you: Starting from scratch?!
 
I'm using "2019-04-08-raspbian-stretch-lite"...
So, you think, too: creating a new OS: 
2022-09-22-raspios-bullseye-armhf-lite.img.xz
This sounds like a long way.
But you convinced me: I'll try
 
Again thanks a lot
Peter

APP (Issue #238)
Datum: 2023-01-09T09:03:01+0100
 
 
 
 

[ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
"https://github.com/bk1285/rpi_wordclock/issues/238#issuecomment-1375231077", 
"url": 
"https://github.com/bk1285/rpi_wordclock/issues/238#issuecomment-1375231077", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"https://github.com" } } ]

FrankX0, 2023-01-09:
I agree this is probably the right time to start fresh and moving to bullseye.
I have my clock running on bullseye, so I expect you will encounter less problems (fingers crossed).

Good luck,
Frank.

phenze, 2023-01-09:
Yes you have to use bullseye (or Buster as a minimum). 
It is not working on stretch.
Maybe we should mention this in the docs @bk1285 

And it is not a long way ;). It is much easier as you might think.

PeterBrenden, 2023-01-09:
Sorry, It's me again, but I think with fairly good news
So far bullseye is done. ssh is working, everything according to https://rpi-wordclock.readthedocs.io/en/main/doc_software_setup.html

Is this right source ? git clone https://github.com/bk1285/rpi_wordclock.git
Because running the wordclock gives an import error:
ImportError: cannot import name 'parse_rule' from 'werkzeug.routing' (/usr/local/lib/python3.9/dist-packages/werkzeug/routing/__init__.py)
I tryed both installing werkzeug 2.1.2 and the betterone werkzeug 2.2.2
Best regards from blackforest
Peter

PeterBrenden, 2023-01-09:
maybe her the complete message:
pi@raspberrypi:~/rpi_wordclock $ sudo python3 wordclock.py
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 17, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 4, in <module>
    from flask_restx import Api, Resource, fields
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/__init__.py", line 5, in <module>
    from .api import Api  # noqa
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/api.py", line 50, in <module>
    from .swagger import Swagger
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/swagger.py", line 18, in <module>
    from werkzeug.routing import parse_rule
ImportError: cannot import name 'parse_rule' from 'werkzeug.routing' (/usr/local/lib/python3.9/dist-packages/werkzeug/routing/__init__.py)

FrankX0, 2023-01-09:
Hi Peter, almost there.....
Can you try:
```
sudo pip3 install flask==2.0.2
sudo pip3 install werkzeug==2.0.2
```

PeterBrenden, 2023-01-09:
Hi Frank,
how did you know?!
Now it is Christmas again. the workclock is running!!!!!!

Now I can start implemention my source.
Thank you so much.
Have a nice evening.
All the best
Peter

FrankX0, 2023-01-10:
Hi Peter,
That is good to hear!
I guess the APP now also works as intended.
If you are extending the functionality you might want to share through a pull request, if you think it could be useful to others.
Kind regards,
Frank.

PeterBrenden, 2023-01-11:
Hi Frank,
yes the APP is running fine!
For now I'm still struggeling with some little challenges, getting my version running on the new plattform (e.g. python2=>3, changes and improvement within your latest wordclock-version).
I'm sure whether my code is that good, some one else would enjoy :) We will see later.
Again many thanks
Peter

PeterBrenden, 2023-01-13:
Hi Frank,
sorry, it's me again. I implented my function, one by one. With last one (sound) the APP worked "for a while", but the browes can't reach it anymore (ssh is still working). Does this error message makes sence to you?

 * Serving Flask app 'wordclock_interfaces.web_interface' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Exception ignored in thread started by: <bound method web_interface.threaded_app of <wordclock_interfaces.web_interface.web_interface object at 0x753d08b0>>
Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 71, in threaded_app
    self.app.run(host='0.0.0.0', port=port)
  File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 920, in run
    run_simple(t.cast(str, host), port, self, **options)
  File "/usr/local/lib/python3.9/dist-packages/werkzeug/serving.py", line 1010, in run_simple
    inner()
  File "/usr/local/lib/python3.9/dist-packages/werkzeug/serving.py", line 962, in inner
    log_startup(srv.socket)
  File "/usr/local/lib/python3.9/dist-packages/werkzeug/serving.py", line 926, in log_startup
    _log("warning", all_addresses_message)
  File "/usr/local/lib/python3.9/dist-packages/werkzeug/_internal.py", line 225, in _log
    getattr(_logger, type)(message.rstrip(), *args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1454, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1583, in _log
    record = self.makeRecord(self.name, level, fn, lno, msg, args,
  File "/usr/lib/python3.9/logging/__init__.py", line 1552, in makeRecord
    rv = _logRecordFactory(name, level, fn, lno, msg, args, exc_info, func,
  File "/usr/lib/python3.9/logging/__init__.py", line 327, in __init__
    self.threadName = threading.current_thread().name
  File "/usr/lib/python3.9/threading.py", line 1328, in current_thread
    return _DummyThread()
  File "/usr/lib/python3.9/threading.py", line 1296, in __init__
    Thread.__init__(self, name=_newname("Dummy-%d"), daemon=True)
  File "/usr/lib/python3.9/threading.py", line 820, in __init__
    self._invoke_excepthook = _make_invoke_excepthook()
  File "/usr/lib/python3.9/threading.py", line 1199, in _make_invoke_excepthook
    raise RuntimeError("sys.excepthook is None")
RuntimeError: sys.excepthook is None

Many thanks from the blackforest
Peter

PeterBrenden, 2023-01-13:
Just in addintion, in case it helps: I did
sudo apt-get install sox
sudo apt-get install libsox-fmt-mp3

PeterBrenden, 2023-01-14:
I think I can stop for now and we could close this issue. I think I have to search for another soundplayer. I'll let you know, when it is hepofully done.

(reasons: with bullsey sox +APP does not work; sox+soundplayer works "sometimes".)

First think for me to do is to learn threading, since I have never done this.

Again many thanks and enjoy programming
Peter

closed 2023-01-14 by PeterBrenden