bk1285/rpi_wordclock #239: No Module named colored logs

basholtman, 2023-01-08:
[image]

What's happening here? Installed everything as described in the manual.

FrankX0, 2023-01-08:
The package "coloredlogs" should have been installed with `sudo pip3 install -r requirements.txt`.
You can install seperately: `sudo pip3 install coloredlogs`
Are you running Python3? Can you run the clock with `sudo python3 wordclock.py`?

basholtman, 2023-01-08:
Hi Frank, 

thanks for the prompt reply. I’ll try sudo pip3 install -r requirements.txt later today. As it wasn’t in the software installation documentation of Bernd. Did I miss more? 

Running python 3.9.2, lite version installed via the raspbian imager. 

Build this project already about 4 years ago, can’t remember above step.

FrankX0, 2023-01-08:
Hi Bas. A lot has changed in 4 years ;-)
This step is described in the [Software setup](https://rpi-wordclock.readthedocs.io/en/main/doc_software_setup.html#software-setup).
Please let me know if you still run into issues.

basholtman, 2023-01-08:
I noticed, getting a bit further now. Still not able to run it due to werkzeug error, how do I read underneath error and correct it?

[image]

FrankX0, 2023-01-08:
Seems related to issue #229 
Can you try:
```
sudo pip3 install flask==2.0.2
sudo pip3 install werkzeug==2.0.2
```

basholtman, 2023-01-09:
[image]

Seems to do the trick, Thank you a lot!

FrankX0, 2023-01-09:
The time is taken from the Raspberry Pi. So my guess is that that time is incorrect.
Can you check that, and if needed, check the timezone in raspi-config?

basholtman, 2023-01-14:
Everything works correctly now!

closed 2023-01-14 by basholtman