bk1285/rpi_wordclock #241: Want to contribute, how to gain access to create branches and submit pull requests for them?

martinvanmiddelkoop, 2023-01-15:
Hi Bernd, I have some ideas to enhance the word clock, and have started developing on my laptop already.
How could I gain access to your repository to contribute?

Some ideas (X = already working):
X Making the simulator flexible for different languages
X Adding a "x-second step-through" option to quickly see all times sequentially for testing (in the simulator most probably)
* adding zones to the word clock for coloring - working on a 5-language word clock + weather station, so 6 sections with each a different coloring
* Adding Chinese (simplified)
* Adding Hindi

mgvanmiddelkoop, 2023-01-15:
I tried with a new personal GIT account, same error. I have a local branch, and want to push it:

```
git -c color.branch=false -c color.diff=false -c color.status=false -c diff.mnemonicprefix=false -c core.quotepath=false -c credential.helper=sourcetree push -v --tags --set-upstream origin refs/heads/Wx-Developer-Simulator-Configurable:refs/heads/Wx-Developer-Simulator-Configurable 
Pushing to https://github.com/bk1285/rpi_wordclock.git
remote: Permission to bk1285/rpi_wordclock.git denied to mgvanmiddelkoop.
fatal: unable to access 'https://github.com/bk1285/rpi_wordclock.git/': The requested URL returned error: 403
Completed with errors, see above
```

phenze, 2023-01-15:
You have to fork the repo and then push the branch to your fork.
Then you can create a pull request

mgvanmiddelkoop, 2023-01-16:
Thanks Phenze, indeed this works! I was used to git and using branches on 1 repository, and was not aware you can actually create a pull request between a forked repository and the "parent" repository.

closed 2023-03-07 by bk1285