bk1285/rpi_wordclock #245: Time no longer updating

R0lleston, 2023-02-09:
Silly me decided to update the buster os on my pi this morning, but now the clock will not update the time.

initially running the word clock.py loads the clock, sets the correct time etc, But after this the time no longer refreshes.

Before the buster package update it was working perfectly 

any ideas.

R0lleston, 2023-02-10:
For us non technical sorts, why would this occur with a working system (not a clean os install) ? Is it the way the word clock has been written or the external programs that interface with it ?

oxivanisher, 2023-02-10:
If my analysis is correct, it is because a newer version of git was installed and this version is now more picky concerning file permissions. But this was on my system after I updated the packages, so it makes sense on my systesm.
Did my solution fix it for you?

FrankX0, 2023-02-12:
@R0lleston can you supply what is being printed in the terminal?

R0lleston, 2023-02-13:
@FrankX0 i think I might have powered down the pi a few too many times without a proper shutdown, I think I’ll need to do a full reinstall of the OS. I’ll let you know when I’ve done that

R0lleston, 2023-02-15:
@FrankX0 have reinstalled buster and loaded the word clock. Can confirm it work’s initially, but soon drops off, no longer updates time and essentially freezes up.

The connection to the app fails, and it longer updates logs when viewed from putty. The pi is still accessible via putty however 

you can restart by the sudo python3 word clock.py commmand

R0lleston, 2023-02-15:
@oxivanisher tried your fix (not as root, but as user pi) worked for a bit but ultimately looked up again

oxivanisher, 2023-02-15:
@R0lleston if you have the same problem as me, you have to do the "fix" (more workaround) as root or it will not help.

R0lleston, 2023-02-15:
@oxivanisher ok, I’ll work out how to do that and give it a crack

oxivanisher, 2023-02-15:
@R0lleston I can help you with that. Connect to the wordclock and run `sudo -i`. After that, you should be working as root. You can exit the root shell again by running `exit`.

FrankX0, 2023-02-15:
@R0lleston I expect the clock application somewhere stops due to an error. Can you supply what is being printed to the terminal?

R0lleston, 2023-02-15:
[image]

R0lleston, 2023-02-15:
@FrankX0 see previous post. As mentioned previously the clock was working fine prior to an update of the Buster OS

FrankX0, 2023-02-15:
This looks fine. There is no information after "Running plugin time_default"? Also not after a few minutes?

R0lleston, 2023-02-15:
Can’t tell the exact time it dies, but it didn’t change any of the data after 3 or 4 mins. Interestingly enough it seems to keep going (more than 5 mins) if the app to change settings is open @FrankX0

R0lleston, 2023-02-15:
Ok confirmed that if I have a connection to the clock via its app it operates fine. I tested it for 1.5hrs with the app open on a chrome tab on my phone, time was constantly correct. Take away that connection and after about 5 mins the clock stops updating and the SSH drops out

FrankX0, 2023-02-15:
It seems to me the raspberry pi loses its WiFi connection.
Can you try the following from a terminal:
sudo iw dev wlan0 set power_save off

R0lleston, 2023-02-15:
@FrankX0 did the above and can confirm the issue remains

FrankX0, 2023-02-16:
I don't see why this issue would be related to the word clock application.
Can you test the network connection without running the word clock application?
Can you try Bullseye instead?

R0lleston, 2023-02-24:
@FrankX0 @oxivanisher looks like I might have found the issue. Seems that the problem occurs when an ssh connection or the app is no longer connected to the pi. A little googling indicates that python programs stop working after ssh disconnects (buggered if I can work out why this has suddenly become an issue with workclock). If you install SCREEN and then run screen python3 rpi_wordclock.py it works as it used to. Screen appears to keep programs running in the background.

FrankX0, 2023-02-25:
This is expected behaviour and not typical for the wordclock.
You might want to start the clock automatically after power on:

- https://rpi-wordclock.readthedocs.io/en/main/doc_software_setup.html#make-software-run-on-every-startup
- Or #246

bk1285, 2023-03-07:
Thanks for looking into this, @FrankX0 @oxivanisher. 

@R0lleston, I close this for now.

Best,
Bernd

HeerNMeester, 2023-03-13:
I had the same problem after a fresh installation. I had to start the clock myself because the cronjob did not work following the ReadtheDocs 'installation' manual. 
After @oxivanisher trick, the clock runs without any issues.