bk1285/rpi_wordclock #247: Wiring for a 16x16 LED matrix

kas1kas, 2023-03-05:
There are a lot of different wiring possibilities to choose from in **wiring.py**. Only one is really documented, the Bends wiring, 
Are there drwaings for these alternative wirings?

Has somebody already written a section to make the workdclock run (more or less) entered on a **16 x 16 LED matrix panel**?

thanks, Kas

maxce-ok, 2023-03-05:
Hi, I programmed a generic 16x16 Display from the shelf based on the logic of this project here for a Raspberry Pi Pico since the big Pi are too expensive and not easy to get.

I am new to this kind of GitHub work but want to share my experience.

I have to modify the original addressing to a WordClock compatible style, see below the addressing conversion for the correct order to work with the first part, an beside that I changed the order of the letters slightly, there is an svg file available for this layout.

# region----------------------------16x16 Definition einer eigenen class
class time_german16:
    """
    This class returns a given time as a range of LED-indices.
    Illuminating these LEDs represents the current time on a german Wordclock
    range 0,2 liefert auf Adresse 0 und 1 also links oben die LEDs für das Wort "ES", etc.
    Purist definiert ob "ES IST" angezeigt wird oder nicht
    Die Grenze für nach liegt bei 24 Minuten also "ZWANZIG NACH" und 4 Punkte für Minuten, danach ist es "FÜNF VOR"
    get_time liefert eine Liste von zu beleuchtenden LEDs zurück als Prefix + Minuten + Stunden + [58] ?? + "Uhr"

    """

    def __init__(self):
        self.prefix = list(range(0, 2)) + list(range(3, 6))
        self.minutes = [[],
                        list(range(27, 31)) + list(range(92, 96)), #FÜNF NACH
                        list(range(34, 38)) + list(range(92, 96)), # ZEHN NACH
                        list(range(72, 79)) + list(range(92, 96)), # VIERTEL NACH
                        list(range(41, 48)) + list(range(92, 96)), # ZWANZIG NACH
                        list(range(27, 31)) + list(range(80, 83)) + list(range(112, 116)), # FÜNF VOR HALB
                        list(range(112, 116)), # HALB
                        list(range(27, 31)) + list(range(92, 96)) + list(range(112, 116)), # FÜNF NACH HALB
                        list(range(41, 48)) + list(range(80, 83)), # ZWANZIG VOR
                        list(range(72, 79)) + list(range(80, 83)), # VIERTEL VOR
                        list(range(34, 38)) + list(range(80, 83)), # ZEHN VOR
                        list(range(27, 31)) + list(range(80, 83))] # FÜNF VOR
        self.hours = [list(range(128, 133)), # ZWÖLF
                      list(range(144, 147)), # EINS
                      list(range(151, 155)), # ZWEI
                      list(range(184, 188)), # DREI
                      list(range(192, 196)), # VIER
                      list(range(139, 143)), # FÜNF
                      list(range(155, 160)), # SECHS
                      list(range(165, 171)), # SIEBEN
                      list(range(179, 183)), # ACHT
                      list(range(118, 122)), # NEUN
                      list(range(160, 164)), # ZEHN
                      list(range(134, 137)), # ELF
                      list(range(128, 133))] # ZWÖLF
        self.full_hour = list(range(221, 224)) # UHR
        self.min_dots = [244, 245, 246, 247] # 1 2 3 4
        self.min_text = list(range(224, 228)) + list(range(249, 255)) # PLUS    MINUTEN

    def get_time(self, hour, minute, purist):
        hour2 = hour % 12 + (1 if minute//5 > 4 else 0) # Stunde geteilt durch 12 (% Modulus) --> Rest + 1 wenn Ganzzahl der Minute /5 >4 sonst 0
        minute2 = minute//5
        minute3 = (minute + 5) % ((minute2 + 1) * 5)
        # Zusammenbau
        # purist for "ES IST" prefix, minutes & hours according lists in class time_german
        return  \
            (self.prefix if not purist else []) + \
            self.minutes[int(minute2)] + \
            self.hours[hour2] + \
            ([147] if (hour2 == 1 and minute2 != 0) else []) + \
            (self.full_hour if (minute2 == 0) else []) + \
            (self.min_dots[int(minute3)-1 : int(minute3)] + \
            self.min_text +\
            ([255] if (minute3 >1) else []) if (minute3 != 0) else [])

# endregion----------------------------16x16 Definition einer eigenen class

# Array 16x16 manuell eingegeben und nicht berechnet
        final_array = array('i', [15,14,13,12,11,10,9,8,7,6,5,4,3,2,1,0,
                                  16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,
                                  47,46,45,44,43,42,41,40,39,38,37,36,35,34,33,32,
                                  48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,
                                  79,78,77,76,75,74,73,72,71,70,69,68,67,66,65,64,
                                  80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,
                                  111,110,109,108,107,106,105,104,103,102,101,100,99,98,97,96,
                                  112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,
                                  143,142,141,140,139,138,137,136,135,134,133,132,131,130,129,128,
                                  144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,
                                  175,174,173,172,171,170,169,168,167,166,165,164,163,162,161,160,
                                  176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,
                                  207,206,205,204,203,202,201,200,199,198,197,196,195,194,193,192,
                                  208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,
                                  239,238,237,236,235,234,233,232,231,230,229,228,227,226,225,224,
                                  240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255])
    return final_array

kas1kas, 2023-03-05:
Nice to see. I don´t think this is directly compatible with the wordclock software, but I will study your solution. Maybe I (others) can use it. The final implemetation should also work with displays other than just the time :-)

maxce-ok, 2023-03-05:
I wanted to use the full blown software but the Pico doesn't support everything.. so I started from scratch and only used the logic from this project, and in addition this is my first sw project over all :-) I am still fighting with the regulations how to present my solution within GitHub without getting troubles about the rules and regulations.