bk1285/rpi_wordclock #254: LEDs not working

MichaelSeitz98, 2023-12-11:
Hi guys, 

Thanks for this cool project. I am (trying to) create a word clock for a surprise Xmas present.

- I have set up the hardware and was able to successfully run the software on the device. 
- However, the LEDs do not light up at all. 
- Sometimes the first two (single, like Bernd's wiring) LEDs will light up, but very erratically, and now not at all. 
- I do not get any error messages from the software side...
- I did not adjust the config file ( in my opinion no need, as same configuration as @bk1285 ) 
- I did not yet build in the buttons as I wanted to test the setup first

Do you experts have any idea what is going on here? Any idea would save me from my desperation 😁
Do I need the buttons to make the functionality work? 

[image]

[image]

Thanks and all the best !
Michael

P.S: falls ihr auf deutsch antworten mögt, geht das auch gerne klar ;)

phenze, 2023-12-11:
Hi,
lies dir mal die 4 Issues durch.

https://github.com/bk1285/rpi_wordclock/issues?q=is%3Aissue+common+ground+

Dein Problem ist dass du kein common ground hast. der Minuspol vom Netzgerät muss mit nem minuspol vom raspberry verbunden sein.

Außerdem sehen deine Lötverbindubgen echt nicht gut aus. Miss da mal durch ob auch überall was anliegt. achte auch darauf dass du keinen kurzschluss hast.

Wenn alles nichts hilft kannst du noch noch in der Mitte von jedem Steifen jeweils den minus und pluspol verbinden

phenze, 2023-12-11:
Kurzschluss merkt man normal schnell.

Die streifen werden dann sehr warm und der raspberry sollte abschalten.

Probier das mit dem strandtest (ist in einem der beiträge erklärt ) mal aus. das ist die einfachste methode zu prüfen ob mit dem led streifen alles passt.

MichaelSeitz98, 2023-12-12:
Hi @phenze 

danke nochmals für deine Antwort!

### Ich hab mittlerweile die folgenden Dinge ausprobiert (Spoiler: leider ohne Erfolg): 

*  versucht den Common Ground zu verbinden: habe den Ground (Schwarz) des Netzteils auf einen Ground des PIs gesteckt. (siehe Bilder).[image]
[image]
 Aber leider ich bekomme dasselbe Resultat: nur die ersten beiden LEDs leuchten und das auch ziemlich willkürlich. Und die 
 reagieren auch manchmal auf Druck/Berührung.  Daher bin ich mir irgendwie nicht mehr sicher, ob ich das richtig gemacht hab. 
 Was sagst du dazu? 
* Auch [diesen](https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/examples/strandtest.py) strandtest.py habe ich gestartet. Das Programm dazu läuft ohne Fehlermeldung, aber auch hier gibt es ein ähnliches Ergebnis: nur die ersten zwei LEDs und willkürlich.  
 Hast du da zufällig auch eine Idee dazu? Wäre dir natürlich um jede Hilfe sehr dankbar!

### Meine nächsten, verzweifelter Versuch wäre der folgende 
1. Ein neues LED-Band kaufen
2. Alles 114 LED am Stück lassen und anschließen (dann muss ich nur 3 Lötstellen setzen und kann schlechte Lötstellen/LEDs ausschließen)
3. Dann nochmals versuchen das Programm zu starten.

Macht der Plan für dich Sinn? 
### Meine verwendete Hardware: 
* Raspberry PI 3 B 
* LEDs: https://www.amazon.de/dp/B09LTT1PBX?psc=1&ref=ppx_yo2ov_dt_b_product_details
* Netzteil: https://www.amazon.de/dp/B08SVX1TH5?ref=ppx_yo2ov_dt_b_product_details&th=1

Besten Dank und schon mal eine schöne Adventszeit!
Michael

phenze, 2023-12-13:
Also schwarz ist dein Datenkanal. Der muss auf Pin 12.
Braun ist dein Ground. Der muss auf Pin 6,9 oder 14. Meiner ist auf 9.

Pinbelegung wie auf dem bild im Anhang.

[image]

Verwende bitte mal das Strandtest Example aus dem Thread hier: https://github.com/bk1285/rpi_wordclock/issues/106

Das ist so umgeschrieben dass es einfach der Reihe nach alle LED's durchschaltet. Ideal zum Testen.

Wenn iwas auf Druck/Bewegung reagiert sieht das sehr nach kalten Lötstellen aus (leider) . Da hilft nur Löten üben. Ein neuer Streifen bringt da meiner Meinung nach nicht viel.
Du kannst auch während der Test läuft einfach mal den Streifen biegen und an den Lötstellen drücken. Dann sieht du den Fehler evtl.

Probier doch erstmal anstatt der einzelnen LED am Anfang einfach einen längeren Streifen an erster Stelle anzulöten und prüfe was dann passiert. Gerade die einzelnen LED's fallen bei mir auch immer mal wieder aus (Bin auch kein Löt Experte).

Hier noch ein Bild wie ich das gelötet habe. Seitdem ich +/- nur in der Mitte verbunden habe und nur den Datenkanal außen funktioniert das bei mir viel stabiler.

[image]

MichaelSeitz98, 2023-12-14:
Hi, 

Danke mal wieder! 

* Dein umgeschriebenen Strandtest.py aus #106 konnte ich nicht 1:1 ausführen, weil ich da einen Error bekommen für Zeile 21: `LED_STRIP = ws.WS2811_STRIP_GR`, weil `ws` nicht bekannt ist. Hab dann `from rpi_ws281x import ws, Color, Adafruit_NeoPixel'  importiert und dann war er aufführbar, aber die komische Effekte bleiben (s. unten). 

* Ich hab jetzt doch einfach mal den neuen Streifen angeschlossen und _Strandtest.py_ aus dem Ordner [example](https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/examples/strandtest.py) (auf 114 LEDs angepasst und nur den ColorWipe dringelassen) ausgeführt. Dann bekomme ich erneut ein sehr komisches Resultat: Es leuchten einige LEDs (siehe Bild), aber nicht alle. Die LEDs wechseln manchmal willkürlich, manchmal auch auf Kontakt.
* Falls das irgendwie noch klappen sollte, schulde ich dir @phenze auf jeden Fall ne Menge!

Hier noch Bilder bzw. Video vom Debakel 🤣 
[image]

https://github.com/bk1285/rpi_wordclock/assets/65539604/a4c29138-1da4-436b-bf48-bc2576ca757e

[image]

https://github.com/bk1285/rpi_wordclock/assets/65539604/424f64a3-d2e6-4ef2-accc-8699d0ee90a7

phenze, 2023-12-14:
Probiere mal bitte noch die anderen konstanten für LED_STRIP aus.

Die findest du hier: https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.h

Ich glaube aber dass bei dir der pluspol einen kurzschluss auf den Datenkanal hat.
Schließe doch einmal +/- eine Ebene weiter hinten an und nur den Datenkanal vorne (wie bei mir auf der Uhr).

Wenn das alles nicht hilft würde ich das System nochmal komplett neu aufsetzen. Sprich Raspbian nochmal komplett neu installieren.

Und wenn das alles nicht hilft weis ich leider auch nicht mehr weiter. Dann kann es euigl nur noch ein defekter Streifen oder Raspberry sein.

phenze, 2023-12-14:
Ahh. Du hast einen Ws2812b und keinen Ws2811b. (12 statt 11). 

Das könnte natürlich auch noch ein Thema sein. Dafür gibt es aber auch Konstanten in dem File oben.
Probier das einfach mal aus.

MichaelSeitz98, 2023-12-14:
Hab mal die Konstanten ausprobiert und vor allem auf die ws2812 umgestellt. Leider auch wieder ohne Erfolg, sondern wieder mit willkürlichen LED-Aktionen beim strandtest. Starte jetzt nochmals von ganz vorne :) wenn das nicht klappt, werde ich wohl einfach ein neuen Raspberry PI versuchen und einen neuen LED Streifen. Wenn das auch nicht hilft, muss ich wohl leider aufgeben 😞 

Könnten die Buttons bzw. die dazugehörigen Widerstände eine Rolle spielen? Soll ich die einfach mal anschließen?

phenze, 2023-12-14:
Nein der Strandtest weiß ja gar nichts von den Buttons. Und auch in der Wordclock Software kannst du diese deaktivieren.

MichaelSeitz98, 2023-12-14:
Danke! Dann werde ich mal einen neuen ED-Streifen und Raspberry Pie versuchen und einen Level Shifter versuchen einzubauen. 
Danke für deine Hilfe!

MichaelSeitz98, 2023-12-15:
Update: scheint als sei der fehler endlich behoben. Es lag tatsächlich am Raspberry Pie!
Ich konnte die LEDs nun anschließen und der strandtest lässt sich endlich mit sinnvollen ergebnissen ausführen. 
Einen kleineren Haken gibt es aber noch: es funktionieren nur die ersten circa 23 LEDs. Ab dann geht der Test nicht weiter. 
Was wäre da deine Einschätzung dazu: den dritten Streifen auszutauschen oder auf deine Verkabelung zu wechseln?

closed 2023-12-16 by MichaelSeitz98