bk1285/rpi_wordclock #256: Auto-startup not working with RPi Zero WH

ghostkills12, 2024-01-19:
Hi,

I've made the wordclock and is fully functional when booted from a laptop via ssh. However, when I change the crontab to get wordclock.py to start automatically on boot nothing happens. I still need to boot the wordclock manually via ssh. I've also tried other ways like cron or systemd service, but I am not familiar with any of those techniques and can't get them to work either.
I am using a Raspberry Pi Zero WH.

MichaelSeitz98, 2024-01-19:
Hi ghostkills12, 

i had similar issues, but I just forgot the "sudo"  in `sudo crontab -e` and `@reboot sudo python3 /home/pi/rpi_wordclock/wordclock.py`. I just activated a "non-sudo" crontab, which resulted in nothing...Did you check that already?

ghostkills12, 2024-01-19:
HI Michael! I made sure to use sudo in crontab (and booting it via ssh), so that is not the issue. Thanks for your quick response though!

Cheers,
Jurre

FrankX0, 2024-01-21:
Please check #172.
Maybe it helps.

ghostkills12, 2024-01-22:
Thanks for the reply FrankXO! Unfortunately, the solution you suggested didn't work. I've looked into the crontab logs, which shows the following:

      git config --global --add safe.directory /home/wordclock/rpi_wordclock
Traceback (most recent call last):
  File "/home/wordclock/rpi_wordclock/wordclock.py", line 201, in <module>
    word_clock = wordclock()
  File "/home/wordclock/rpi_wordclock/wordclock.py", line 31, in __init__
    self.currentGitHash = subprocess.check_output(["git", "describe", "--tags"], cwd=self.basePath).strip().decode()
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'describe', '--tags']' returned non-zero exit status 128.

FrankX0, 2024-01-22:
It looks like the code is trying to read the version from the git information, but it cannot find it (exit code 128).
Can you comment-out lines 31 and 32 in wordclock.py and try again?

ghostkills12, 2024-01-22:
That did the trick! Awesome FrankXO, your help is much appreciated :D

oxivanisher, 2024-01-22:
This will happen more and more to all new installs. Please see this PR with a fix: https://github.com/bk1285/rpi_wordclock/pull/250

You problem is with the new git version and a "not so clean way" to start the wordclock trough cron as root when it gets installed as user. See my comments in https://github.com/bk1285/rpi_wordclock/issues/245, they will help you with your current problem.

ghostkills12, 2024-01-22:
Thank you for the tip oxivanisher. I completely agree that running the wordclock as root is not a clean way and  that the issues with this quick fix may rise again in future updates. I'll have a good look on those PR's tomorrow!