bk1285/rpi_wordclock #258: No module named 'flask.scaffold'

teletext78, 2024-06-18:
Hello,

When running the software for the first time, I get this error:

pi@wordclock:~/rpi_wordclock/wordclock_config $ cd ~/rpi_wordclock
sudo python3 wordclock.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/api.py", line 21, in            <module>
    from flask.helpers import _endpoint_from_view_func
ImportError: cannot import name '_endpoint_from_view_func' from 'flask.helpers'            (/usr/local/lib/python3.9/dist-packages/flask/helpers.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/rpi_wordclock/wordclock.py", line 17, in <module>
    import wordclock_interfaces.web_interface as wciweb
  File "/home/pi/rpi_wordclock/wordclock_interfaces/web_interface.py", line 4, i           n <module>
    from flask_restx import Api, Resource, fields
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/__init__.py", line 5,            in <module>
    from .api import Api  # noqa
  File "/usr/local/lib/python3.9/dist-packages/flask_restx/api.py", line 23, in            <module>
    from flask.scaffold import _endpoint_from_view_func
ModuleNotFoundError: No module named 'flask.scaffold'

Any idea?

closed 2024-06-20 by teletext78

teletext78, 2024-06-20:
Problem solved:
I've installed Raspbian Bullseye lite on a RPi 1B using Raspberry Pi Imager.
To fix my issue, I had to install the following:

sudo pip3 install flask==2.0.2
sudo pip3 install werkzeug==2.0.2

Sidenote: 
with Raspbian Bookworm (latest distro at the moment of writting), when trying to run "sudo pip3 install", I always get the error message "error: externally-managed-environment", that's why I've installed Bullseye instead.

Hope this will help somebody else as well.

oxivanisher, 2024-06-20:
Concerning the sidenote: My PR #250 would solve this issue.