bk1285/rpi_wordclock #259: Wie sehen die Wirings aus?

coolio210, 2024-09-12:
Hallo, 
mein LED-Strip passt nicht ideal, wenn ich den Verlauf nach "bernds_wiring" installiere (ich muss die Stripes pro Spalte zweimal trennen und jeweils minimal verlängern, damit es genau passt).
Ein horizontaler LED-Verlauf würde exakt passen. 
Deshalb meine Frage: 
Sind die in der Datei  wiring.py erwähnten wiring-Layouts "christians_wiring" bzw. timos_wiring irgendwo dokumentiert, dass man deren Verlauf erkennen und entsprechend nachbilden kann?
Danke schon mal und dickes Lob für dieses tolle Projekt.

bhunziker, 2024-09-14:
Hallo coolio210
Ich hatte eine ähnliche Aufgabe, habe immer 2 LED pro Buchstabe. Mit einer angepassten dein_wiring Klasse im wiring.py kannst du dies definieren. Beispiele hat es ja viele in diesem file.
Ich habe zuerst in Excel die Formel für die Umrechnung der 2d Koordinaten auf die LED Anordnung entwickelt und dann codiert.
Gruss Bernhard

bhunziker, 2024-10-06:
Ein Beispiel mit Verdrahtung ist auf https://www.mikrocontroller.net/articles/WordClock_mit_WS2812#Anschluss_WS2812-Streifen_f.C3.BCr_WordClock12h dokumentiert.
Du kannst auch mit vielen if-then-else die Aufgabe lösen. Dies ist nicht so elegant, kommt aber komplett ohne Mathematik aus.
Meine Lösung mit 2 LED pro Buchstabe von oben links jeweils nach unten verdrahtet gab folgenden Code:
        # WENN(REST(x,2)=0,x*height+y,(x+1)*height-y-1)
        if x % 2 == 0:
            return x * self.WCA_HEIGHT + y
        else:
            return (x + 1) * self.WCA_HEIGHT - y - 1